adafruit/circuitpython #2059: Changes to displayio caused SSD1351 to stop working with CircuitPython 4.1.0

makermelissa, 2019-08-13:
It appears some of the displayio changes in 4.1.0 may have broken the SSD1351 driver. I just tested that the driver works fine in 4.0.1 and I will see if I can narrow down further with which iteration it actually stopped working. I'm wondering if it had anything to do with us removing that delay.

makermelissa, 2019-08-13:
Ok, it seem to have broken between 4.1.0 Beta 0 (working) and 4.1.0 Beta 1 (not working).

makermelissa, 2019-08-13:
The change was that the baudrate was added as a parameter to Fourwire and the default speed was increased. Manually passing in 16000000 for the baudrate has it working in 4.1.0.

closed 2019-08-13 by makermelissa

tannewt, 2019-08-13:
Is 16mhz the limit of the ssd1351 or of the wiring? If it's the chip, we should add a way for the driver to reduce the baudrate of the given bus.

makermelissa, 2019-08-13:
Yes, that seems to be the limit of the SSD1351. Currently the driver is separate from the bus. We could do a driver sweep and add that to the driver and reversion them as 2.0. For now though, I'm working on the guide page and I will be putting that in there.

makermelissa, 2019-08-13:
I'll also update the example.

ladyada, 2019-08-13:
...technically.... many our displays have SPI limits, we just ignore them because they dont mind overclocking. the HX for example, doesn't like going over 32MHz, but the ST77 is happy up to 80MHz+!

makermelissa, 2019-08-13:
Well, if I set it to 20MHz, it doesn't work.

tannewt, 2019-08-13:
I'm thinking that we can expose a baudrate property on the bus and have the driver cap it to the max it can handle.

makermelissa, 2019-08-13:
Can it be changed once the Fourwire object is declared?

makermelissa, 2019-08-13:
Nevermind, I just reread what you typed. That's what you're proposing. Sounds good.

deshipu, 2019-08-14:
> I'm thinking that we can expose a baudrate property on the bus and have the driver cap it to the max it can handle.

That sounds like a magical side effect that will only cause problems at some point to someone. I know it's not as user-friendly, but perhaps just checking the baud rate and raising an exception when it's too high would be better?

tannewt, 2019-08-16:
Sure, throwing an exception works too.

makermelissa, 2019-08-16:
We would still need to expose the property so the driver can check what it’s set to. Also it would need to fail gracefully if used with an older version of CircuitPython.