adafruit/circuitpython #2060: I2C error after soft reboot

jerryneedell, 2019-08-13:
I get the following error accessing an I2C device  (VCNL4040) after a soft reboot.
It  works normally after a hard reboot, but fails after a soft reboot (control-D)

This occurs with CP4.1.0 and CP 5.0 (current master)
 

```
HARD RESET
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.1.0 on 2019-08-03; SparkFun Pro nRF52840 Mini with nRF52840
>>> import vcnl4040_simpletest
Proximity: 1
Light: 824
White: 2359
Proximity: 1
Light: 825
White: 2372
Proximity: 1
Light: 830
White: 2385
Proximity: 1
Light: 827
White: 2398
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "vcnl4040_simpletest.py", line 13, in <module>
KeyboardInterrupt: 
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.1.0 on 2019-08-03; SparkFun Pro nRF52840 Mini with nRF52840
>>> import vcnl4040_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "vcnl4040_simpletest.py", line 7, in <module>
  File "adafruit_vcnl4040.py", line 277, in __init__
  File "adafruit_register/i2c_struct.py", line 86, in __get__
  File "adafruit_register/i2c_struct.py", line 86, in __get__
  File "adafruit_bus_device/i2c_device.py", line 154, in write_then_readinto
  File "adafruit_bus_device/i2c_device.py", line 96, in readinto
OSError: [Errno 19] Unsupported operation
>>>

jerryneedell, 2019-08-13:
I am not able to reproduce this error on a SAMD51 (metro_m4_airlift_lite) it appears to be nRF specific.

jepler, 2019-08-13:
I will try to reproduce this on a Particle Xenon with OLED featherwing, I believe that has an I2C interface.

I did verify just reading the source that reset_port() calls i2c_reset(), so it's not the most obvious possible cause.

jerryneedell, 2019-08-13:
I just reproduced it on a Particle Xenon as well - I have not tried the I2C devices -- will do that next
So far I have just used the vcnl4040.

dhalbert, 2019-08-13:
@jerryneedell Could you try something besides a VCNL4040 or VEML7700? These boards have currently have pullups that are too weak for the chips (which need 4.7k instead of 10k for reliable operation).

jerryneedell, 2019-08-13:
SO far I cannot reproduce it with anything else -- I have tried am2320, adxl34x and adt7410
all work fine  -- may well just be the vcnl4040

jerryneedell, 2019-08-13:
why is it just impacting the nrf and not SAMD?

dhalbert, 2019-08-13:
Feather '840 has very short traces to I2C pins, shorter than other Feathers. Capacitance isn't quite enough. Adding 10pF caps or extra 10k pullups makes those chips work fine on Feather '840, but they don't work properly otherwise. Datasheet says 2.2k or 2.2k-4.7k for pullups. This is lower than most pullups. Metro '840 proto works with these chips with current 10k pullups. Try adding extra 10k pullups from 3.3V to ground on the SDA/SCL pins and see if this issue is still present. We're revising the boards.

jerryneedell, 2019-08-13:
hmm - I have 2 other devices with 10K pull-ups on the bus I thought that would help but I'll try adding them directly

dhalbert, 2019-08-13:
It may be a different problem.

jerryneedell, 2019-08-13:
also, why would it only fail after reboot but work after hard reset?

dhalbert, 2019-08-13:
I saw the TWIM (I2C) peripherals seem get into a bad state while debugging the VCNL/VEML problem. Only a hard reset would fix them. But there may be some incomplete resetting going on. If we can duplicate the problem on other I2C devices, then we can eliminate chip-specific problem and it's a software problem, not a hardware problem.

jerryneedell, 2019-08-13:
adding 10K pull-ups does not help :-(

deshipu, 2019-08-13:
4.7kΩ are standard for high-speed I²C

jerryneedell, 2019-08-13:
FWIW - I accessed an adxl34x twice -- hard reset then soft reset -- no problem 
then did a soft reset and tried to access the vcnl4040 and it failed as above.

jerryneedell, 2019-08-13:
@dhalbert did you really mean 

> Try adding extra 10k pullups from 3.3V to ground on the SDA/SCL pins

I have them from SCA/SDL to 3.3v  -- not ground

dhalbert, 2019-08-13:
No, I didn't, sorry, doing multiple things at once :)

jerryneedell, 2019-08-13:
also tried adding 10pF cap between 3.3V and GND but it did not change anything.

jerryneedell, 2019-08-13:
So - at this point - this does appear to be an issue that only impact the vcnl4040 -- the only reason I stumbled across it is taht I wanted to try the qwiic connectors on the sparkfun nrf52840 and daisy chain a few qwic compatible sensors.  That test was successful...as long as don't do a soft reboot!  Would you prefer to move this to the vcnl4040 driver repo?

jepler, 2019-08-13:
Based on further findings, I'm deassigning myself.  I'll still check with the i2c part I have, just in case, but it sounds like I won't be able to repro the problem

jerryneedell, 2019-08-13:
FYI -- the error also occurs, as predicted, with the VEML7700...

dhalbert, 2020-02-03:
Closing this because it was really an issue with the Vishay parts needing stronger pullups on their boards (4.7k vs typical 10k). Those breakouts have been revised to include more suitable resistors.

closed 2020-02-03 by dhalbert