adafruit/circuitpython #2076: Transparency setting for shape fill color cannot be dynamically changed after initialization

jpecor, 2019-08-18:
I ran into some issues trying to switch between transparent and colored fills for some buttons I was using in a recent project.

Once a shape is created with either fill=<SOME_COLOR> or fill=None, the shape will act as either a filled **OR** transparent shape and cannot be changed.

The issue appears to be related to the Palette core module in displayio, and the symptoms show up in Adafruit_Display_Shapes as well as Adafruit_Display_Buttons.

It seems like the intent is to allow shapes to change from a solid color fill to transparent and vice-versa, but it doesn't work.

I created a short program that provides a simple demo/test that shows the failure. By selecting whether or not the starting fill is a color (solid, opaque color) or None (transparent), the related failure can be seen.

The test is here: 
https://github.com/jpecor/circuitpython_stuff/blob/master/fill_color_none_test.py

In the test, if the starting shape starts with a color, the program will cycle through a few different colors and then attempt to make the shape transparent which will not work.

If the starting shape is transparent (fill=None), the program will again cycle through different colors which will have NO effect as expected, but then it will attempt to change the body color to an actual fill color. This should work, but it does not.

I believe the bug itself is buried somewhere in the Palette module of displayio, but I don't fully understand how the core all stitches together or works at this point.

Maybe someone with more experience can have a look?

jpecor, 2019-08-19:
Wait, actually, the initial attempts to change the fill color when the shape is transparent should work, as well.  Not sure what I was thinking there.  It's the same function call.  Sigh...

tannewt, 2019-08-20:
Did the modified version work? I was about to suggest that. 

Looks like make_opaque and make_transparent don't trigger a refresh on their own: https://github.com/adafruit/circuitpython/blob/master/shared-module/displayio/Palette.c#L36

jpecor, 2019-08-20:
No, the modified version does not work, either.

jpecor, 2019-09-13:
I took a crack at adding "self->needs_refresh = true;" to both common_hal_displayio_palette_make_opaque() and common_hal_displayio_palette_make_transparent() in Palette.c.  Did not change the behavior I am seeing.  

Any suggestions?  Interested in continuing to chase this, but open to some guidance.

tannewt, 2019-09-14:
@jpecor thanks for taking a crack at it! Is the edited version available in a branch somewhere? I'd be curious to see what you did. The next place to look is here: https://github.com/adafruit/circuitpython/blob/master/shared-module/displayio/TileGrid.c#L550 That is how palette state gets propagated up into TileGrid and then Group.

jpecor, 2019-09-14:
@tannewt   - I don't have it committed anywhere yet.  I confess that I'm in a bit over my head with remote repos and how to create a branch remotely.  So, I'm a bit hesitant there.  

I just set needs_refresh = true in both make_opaque() and make_transparent():  

```
void common_hal_displayio_palette_make_opaque(displayio_palette_t* self, uint32_t palette_index) {
    self->colors[palette_index].transparent = false;
    self->needs_refresh = true;
}

void common_hal_displayio_palette_make_transparent(displayio_palette_t* self, uint32_t palette_index) {
    self->colors[palette_index].transparent = true;
    self->needs_refresh = true;
}
```

tannewt, 2019-09-16:
@jpecor Let's chat for an hour or so sometime this week. We can talk displayio and git. What times work for you this week? (Feel free to DM on Discord your reply.)

DavePutz, 2019-10-30:
This issue looks to be in the library. See https://github.com/adafruit/Adafruit_CircuitPython_Display_Shapes/issues/7
for details.

makermelissa, 2019-11-01:
Fixed in adafruit/Adafruit_CircuitPython_Display_Shapes#8

closed 2019-11-01 by makermelissa