adafruit/circuitpython #2078: nrf: pwmaudioout: RawSample always loops

jepler, 2019-08-19:
It's some kind of problem with the way I'm using the nrf's PWM module.  It may also affect short WaveFiles; I haven't fully characterized it yet.  If I don't get it solved soon, I'll add a reproducer script here in case somebody else wants to take a crack at it.

kevinjwalters, 2019-12-16:
I've been playing wave files ok using `cpb.play_file()`. The standard sample code reproduces this in REPL on my CPB:

```
import time
import array
import math
import board
import digitalio
from audiocore import RawSample
from audiopwmio import PWMAudioOut as AudioOut

speaker_enable = digitalio.DigitalInOut(board.SPEAKER_ENABLE)
speaker_enable.direction = digitalio.Direction.OUTPUT
speaker_enable.value = True

audio = AudioOut(board.SPEAKER)

length = 8000 // 440
print(8000 / length)

sine_wave = array.array("H", [0] * length)
for i in range(length):
    sine_wave[i] = int(math.sin(math.pi * 2 * i / 18) * 32767 + 32768)

sine_wave_sample = RawSample(sine_wave)

audio.play(sine_wave_sample, loop=False) ; time.sleep(3) ; audio.stop()
```

You'd expect that to make a probably inaudible noise for about 2ms but it plays for _three seconds_. That would be expected for `loop=True` as previously noted in discord circuitpython discussion.

kattni, 2019-12-16:
I was able to reproduce this as well. It was also reported by a user with regards to Circuit Playground Express code used on the Circuit Playground Bluefruit: https://github.com/adafruit/Adafruit_Learning_System_Guides/issues/970

kattni, 2019-12-16:
@jepler Please bump this up the priority list. Spend this week looking into it. If it turns out to take longer than that, we'll move it to a long-term bug list at that point. Thanks!

jepler, 2019-12-17:
It'll be my focus starting tomorrow, I ended up working in displayio stuff today

jepler, 2019-12-17:
Closed by #2400

closed 2019-12-17 by jepler