adafruit/circuitpython #2082: Remove stop kwarg from busio.I2C().writeto

tannewt, 2019-08-21:
With the introduction of `writeto_then_readfrom` into busio.I2C in 5.x. The stop kwarg is no longer needed. In fact, using it is likely the wrong thing to do since it is unsupported with Blinka.

kattni, 2019-08-22:
We're looking for any assistance in testing all of the outstanding library PRs associated with this issue! 

If you have any of these sensors and would like to help out, please grab something like a Feather M4 Express and do the following:

1. Update to the latest Adafruit CircuitPython BusDevice release (4.0.0, released 22 August 2019)
2. Find the PR for the sensor you'd like to test.
3. Comment on the PR that you are going to test the PR. (This will avoid any duplicate work!)
4. Test the code from the PR.
5. If tested successfully, approve and merge the PR. If you do not have access to merge, please comment on the PR that you have tested it successfully and tag me (@kattni) in the approval comment.
6. Do a new release on the library. (Again, if you do not have access, please comment on the PR and tag me (@kattni) in the comment.

If you have any questions, please tag me in a comment or find me on [Discord](https://adafru.it/discord) in the #circuitpython channel.

Thanks!

tampe125, 2019-08-29:
Hello, this change send to have broken some stuff. More info here https://github.com/adafruit/Adafruit_CircuitPython_VL6180X/issues/7

closed 2020-07-23 by tannewt

kevinjwalters, 2020-12-09:
This has broken:

- https://github.com/pimoroni/circuitpython_adapter/blob/15751dcdce546a5382d442ebf344742adf76f9eb/library/circuitpython_adapter/__init__.py#L36..L46 - I'll discuss this with the Pirates
- and possibly https://github.com/adafruit/glitterpos which is rather unusual in having old copies of libraries in `lib` - I put feedback in Guide system to check this.

PR #3195 (2020-07-23): Remove stop kwarg from I2C writeto.
Fixes #2082
commit 4b6e0294 (2020-07-23): Remove stop kwarg from I2C writeto.

Fixes #2082