adafruit/circuitpython #2086: PWMOut failed when changing frequency in an infinite loop

huynh213, 2019-08-23:
I try to understand why it failed to generate PWM signals after 30+ seconds or so on my ATSAMD21G18. Other functions (such as print) seem to work just fine but just the PWM function stop generate PWM signals. 

I did test it when changing duty_cycle in an infinite loop, and it works just fine. Not sure why it causes a problem with switching frequency. Any thought?

The code is pretty simple as:    
import time
import board
import pulseio

PIN_Test = pulseio.PWMOut(board.A4, frequency=5000, duty_cycle=2 ** 15,  variable_frequency=True)

while True:
    FRE = random.randint(1000, 2000)
    while COUNT >0:
        COUNT -=1
        PIN_Test.frequency=FRE
        print("NEW COUNT = ", COUNT, "at FRE =", FRE)

jepler, 2019-08-23:
Thank you for your report.

I believe I have reproduced the problem.  I happen to be testing with ```Adafruit CircuitPython 3.0.3 on 2018-10-10; Adafruit Feather M0 Express with samd21g18``` and I used a modified version of your test program:
```
import time 
import board 
import pulseio 
import random 
 
PIN_Test = pulseio.PWMOut(board.D5, frequency=5000, duty_cycle=2 ** 15,  variable_frequency=True) 
 
while True: 
    FRE = random.randint(100, 200) 
    COUNT = random.randint(1, 20) 
    print(FRE) 
    while COUNT >0: 
        COUNT -=1 
        PIN_Test.frequency=FRE
        #PIN_Test.duty_cycle = 2**15
        print(PIN_Test.duty_cycle)
```
My initial findings are that when the duty_cycle pin is NOT assigned a value, the actual duty cycle tends to decrease over time (a fact that can be verified by oscilloscope or probably just by attaching an LED), and eventually falls to zero.  By uncommenting the line that assigns to duty_cycle, I can mask the problem.

While it appears that this is a circuitpython bug that should be addressed, you might try just adding an assignment of duty_cycle each time you update frequency.  This may permit you to work around the bug.

jepler, 2019-08-23:
I also verified that this bug exists in a recent development version of circuitpython, `Adafruit CircuitPython 5.0.0-alpha.0-176-g912fd7759 on 2019-08-23; Adafruit Feather M0 Express with samd21g18`

huynh213, 2019-08-23:
jepler,

You're absolutely correct. The duty-cycle decreases over time to the point it becomes 0.
Your way of assigning a new value of duty cycle every loop seems to resolve the issue.

tannewt, 2019-08-30:
Internally, duty_cycle is less than 16 bits because of the top value varies with frequency. As a result, the duty cycle gets recomputed when the frequency is set. Maybe we want to maintain the original duty cycle internally to guard against this.

PR #2240 (2019-10-28): Track unadjusted PWM duty cycle to avoid accumulating conversion errors
Fixes #2086

When the frequency of a `PWMOut` is changed it re-sets the PWM's duty cycle as
well, since the registers have to be re-calculated based on the new frequency.
Unfortunately, `common_hal_pulseio_pwmout_get_duty_cycle`
will return a value very close to, but not exactly, the value passed to `common_hal_pulseio_pwmout_set_duty_cycle`. If the frequency is modified
without the calling code also re-setting the duty cycle then the duty cycle
will decay over time. This fixes that problem by tracking the unadjusted duty
 [... cut]
commit 4fdf5182 (2019-10-28): Track unadjusted PWM duty cycle to avoid accumulating conversion errors

Fixes #2086

When the frequency of a `PWMOut` is change it re-sets the PWM's duty cycle as
well, since the registers have to be re-calculated based on the new frequency.
Unfortunately, `common_hal_pulseio_pwmout_get_duty_cycle`
will return a value very close to, but not exactly, the value passed to `common_hal_pulseio_pwmout_set_duty_cycle`. If the frequency is modified
without the calling code also re-setting the duty cycle then the duty cycle
will decay over time. This fixes that problem by tracking the unadjusted duty
 [... cut]
closed 2019-10-28 by dhalbert