adafruit/circuitpython #2097: socket.settimeout() raising OSError

brentru, 2019-08-30:
Setting `socket.timeout()` to any integer value causes an `OSError: [Errno 22]`. The timeout `value` parameter specified in the [documentation](https://circuitpython.readthedocs.io/en/latest/shared-bindings/socket/__init__.html#socket.settimeout) for the socket module states that it accepts an integer argument, even with setting a timeout of 0 for non-blocking sockets.

@nickzoic  Any idea why this would be occuring?

Code:
[EthernetManager](https://github.com/brentru/Adafruit_CircuitPython_EthernetManager/blob/master/adafruit_ethernetmanager.py) for replication.
```
import board
import busio
import time
import neopixel

import socket
import adafruit_ethernetmanager as eth
import adafruit_requests as requests

spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
status_light = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.2)

# Initialize a new EthernetManager object
ethernet = eth.EthernetManager(spi, board.D10, status_pixel=status_light, debug=True)

# Attempt to connect the ethernet hardware
print("Connecting...")
ethernet.connect()
print("Connected with IP Address: ", ethernet.ip_address)

# Create a new socket object
sock = socket.socket()

addr_info = (2, 1, 0, '', ('104.236.193.178', 80))
sock.connect(addr_info[-1])
sock.settimeout(1)
```

Output from a `Adafruit CircuitPython 4.1.0 on 2019-08-02; Adafruit Feather M4 Express with samd51j19`
```
....
Connected with IP Address:  SNIP
Traceback (most recent call last):
  File "code.py", line 26, in <module>
OSError: [Errno 22] Invalid argument
```

brentru, 2019-08-30:
This seems similar to https://github.com/adafruit/circuitpython/issues/2073

nickzoic, 2020-01-21:
Oh, you mean *this* code, copied straight from MicroPython:
```
    // TODO
    *_errno = MP_EINVAL;
    return -1;
```
Yeah, I should fix that. Unfortunately the Wiznet5500 doesn't seem to support a socket timeout, just a "sock_io_mode" bitmask so each socket is either blocking (polled) or non-blocking (returns SOCK_BUSY) ...

brentru, 2020-01-22:
@nickzoic  Thanks for picking this back up, I'm excited to see progress on ethernet again (as I'd love to write a `EthernetManager` for CircuitPython's `requests` module). Please tag me in the pull request so I can test it on my hardware.

nickzoic, 2020-01-24:
No worries, will do.  Because the W5500 library doesn't really have timeouts per se it'll need to be implemented as polling in our code, which is kind of awful.

nickzoic, 2020-01-30:
OK so that code is now doing some timeout-like behaviour :-) but I need to do some thorough testing,

hierophect, 2021-02-10:
@tannewt @brentru is this issue still relevant with the changes to Socket?

brentru, 2021-02-11:
@hierophect  Don't know - I've moved to using the pure-python implementation of this driver instead.... https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/

hierophect, 2021-02-11:
Ok. Since this is based on an out of date version of socket, and the driver is no longer even using it, I think this is safe to close.

closed 2021-02-11 by hierophect