adafruit/circuitpython #2098: SSD1306 hangs with 5.0alpha

jerryneedell, 2019-08-30:
This code causes the board to hang -- run with feather_nrf52840_express and OLED feathering. It was working prior to alpha release

```
import board
import displayio
import adafruit_displayio_ssd1306

displayio.release_displays()

i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3c)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)
```

anecdata, 2019-08-31:
Only clue I can add is that I reported on Discord on 8/26 that the latest 5.x build was working fine with OLED FeatherWing... I believe that would have been the 8/22 build 1d973a0. But 5.0 alpha stops user code from starting up if OLED is attached: no serial/REPL/^C. Solid green LED on M4.

anecdata, 2019-08-31:
8/22 build 1d973a0 re-confirmed works with OLED.
Subsequent builds all have 5.0 alpha behavior.

Between each of the following, I reloaded the working 8/22 build 1d973a0:

8/27 build 42956a6 : user code runs to a point, OLED gets blanked, then hung
8/28 build 7cbae3d : user code runs to a point, OLED gets blanked, then hung
8/29 build 193df55 : user code runs to a point, OLED gets blanked, then hung

8/29 build 8d14266 : doesn't blank the prior OLED display, then runs up to a point, but with no new OLED output, then:
```
display_bus = displayio.I2CDisplay(i2c, device_address=0x3c)
AttributeError: 'module' object has no attribute 'I2CDisplay'
```

8/29 build 2b5ce66 : user code runs to a point, OLED gets blanked, then hung

8/28 build 8e14522 : doesn't blank the prior OLED display, then runs up to a point, but with no new OLED output, then:
```
display_bus = displayio.I2CDisplay(i2c, device_address=0x3c)
AttributeError: 'module' object has no attribute 'I2CDisplay'
```

and finally...

8/30 build b954b2f. : user code runs to a point, OLED gets blanked, then hung

That's the timeline from the working build to the alpha from S3. Other than the missing attribute, sems to be the same basic behavior introduced in the 8/27 build 42956a6

jerryneedell, 2019-08-31:
I connected a J-Link to the feather_nrf52840 and OLED -- and rebuilt 5.0alpha with DEBUG=1
The code appers to be executing, that is, it does not hard Fault but the display is blank and
it won't respond to any keyboard input (COntrol-C) does not work
Here is a screenshol of the call stack after loading the oled.py test progrram
[image]

this was the REPL
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.1 on 2019-08-31; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> import oled

Hung -- no response
```

PR #2111 (2019-09-03): Fix I2CDisplay bus_free to not grab lock
Fixes #2098
commit b53f1698 (2019-09-03): Fix I2CDisplay bus_free to not grab lock

Fixes #2098
closed 2019-09-03 by dhalbert