adafruit/circuitpython #2099: Another stereo audio issue on M4

jepler, 2019-08-31:
Testing setup: A0, A1 into o'scope.  Put some.wav and code.py on CIRCUITPY.

Initially, A0 and A1 both display a signal, but after a short time (2-3s?), A1 stops showing signal.  At next soft reboot, A1 gets a signal for the same length of time again.

Tested in 4.1.0 and master on a Metro M4 Express.

Code to create wave file: 
```
import array
from numpy import ndarray
import scipy.io.wavfile

X=4000
Y=65535

a = ndarray((16384, 2), 'h')
print(a)
for i in range(0,16384,2):
    a[i,0] = X
    a[i,1] = X
scipy.io.wavfile.write("some.wav", 100000, a)
```

Code to run on circuitpython:
```
import board
import audioio
import time

some_wav = open('some.wav', 'rb')
sample = audioio.WaveFile(some_wav)

a = audioio.AudioOut(board.A0, right_channel=board.A1)

a.play(sample, loop=True)

while 1:
    time.sleep(1)
```

dhalbert, 2019-08-31:
This sounds related to the "refresh" needed on the SAMD51 DAC's. Their output needs to be refreshed explicitly by forcing a reconversion, or else the charge will decay away. If you put various pulldowns on A1 I have a hunch you'll see it decay faster depending on the value of the pulldown.

Turning on refresh is done in `ports/atmel_samd/asf4_conf/hpl_dac_config.h`: `CONF_DAC0_REFRESH` and `CONF_DAC1_REFRESH` are both set to 2. But the symptoms above sound like that value is being changed. Maybe the audio code is somehow ignoring that setting or overriding it?

I originally fixed this issue in https://github.com/adafruit/circuitpython/pull/649. @kevinjwalters mentioned this also in https://github.com/adafruit/circuitpython/issues/1991#issue-466267344.

jepler, 2019-08-31:
I will check into that, but doesn't this apply only to static signals?  The value is being changed a nominal 100,000 times per second between 0 to 4000 continuously.

dhalbert, 2019-08-31:
Good point. Is the loss of signal abrupt or does it fade down? Abrupt would sound like a DMA problem (if DMA is being used here).

jepler, 2019-08-31:
abrupt, like DMA stops

kevinjwalters, 2019-08-31:
There's also another setting about current (`CONF_DAC0_CCTRL` and `CONF_DAC1_CCTRL`). I've not experimented with this, just came across it in the MicroChip docs. Mentioned on: [Adafruit Forums: What's fastest practical sample_rate for M4 DACs?
](https://forums.adafruit.com/viewtopic.php?f=60&t=153797).

jepler, 2019-08-31:
I updated the initial comment with new code and an oscope grab.  And just when you got things weird enough, I got  this one too:
[image]

jepler, 2019-08-31:
[image]

kevinjwalters, 2019-09-01:
I think the SAMD51 has a pseudo-differential mode for the DAC pair? When I saw that in docs I realised bugs around that area could be a cause of _som_e of these coupling-related oddities but there seem to be so many strange quirks that defy an obvious explanation. I did a quick check of the SAMD21 on a CPX a few months back and it's so clean and predictable! It's as if there's a poltergeist in the SAMD51's DACs.

kevinjwalters, 2019-09-01:
I wonder if we should take [Malvin's advice](https://www.youtube.com/watch?v=GfJJk7i0NTk&t=2m42sl) and start at the beginning? Even at first use the SAMD51 DACs used from CircuitPython behave strangely with #1991 .

jepler, 2019-09-02:
Based on this looking "right" in an Arduino environment, I am hoping there's just some spot of setup that we've missed.  Unfortunately, I think I've ruled out everything in the DAC's registers.  Moving on to clock setup now, h/t @sommersoft

kevinjwalters, 2019-09-02:
You are likely to have already observed it or read it, but when I last looked it wasn't clocked fast enough to get the higher rates, 3rd post in https://forums.adafruit.com/viewtopic.php?f=60&t=153797#p759051

jepler, 2020-05-15:
Since #2552 we use just a single DMA channel to send stereo audio (at least when it's 16 bits).  This should resolve such problems.  If the problem is not resolved with CircuitPython 5.3.0, feel free to reopen this bug.

closed 2020-05-15 by jepler