adafruit/circuitpython #2110: NeoTrellis example code crashing on trellis.sync() call

jedgarpark, 2019-09-03:
Hooked up a NeoTrellis board to a Feather M4, tried basic NeoTrellis example code found here: https://learn.adafruit.com/adafruit-neotrellis/circuitpython-code
Code runs through the light test, then hangs, or allows a couple button presses before hanging.
Error:
```code.py output:
Traceback (most recent call last):
  File "code.py", line 49, in <module>
  File "adafruit_neotrellis/neotrellis.py", line 89, in sync
  File "adafruit_seesaw/keypad.py", line 74, in count
  File "adafruit_seesaw/seesaw.py", line 372, in read8
  File "adafruit_seesaw/seesaw.py", line 383, in read
  File "adafruit_seesaw/seesaw.py", line 383, in read
  File "adafruit_bus_device/i2c_device.py", line 96, in readinto
OSError: [Errno 5] Input/output error```

This is running 4.1.0 release CircuitPython and associated bundle libraries for seesaw, neotrellis, and bus_device.

Note: code runs properly under 3.1.2 circuitpython, buttons light up when pressed, no crashing.

makermelissa, 2019-09-15:
I experienced the same thing. The latest version of CP that I could get working with this was 4.0.2. It seems to break after the speedups.

ladyada, 2019-09-15:
@makermelissa @jedgarpark please update the code to add a delay so that it works with 5

makermelissa, 2019-09-15:
Makes sense. 👍🏻

makermelissa, 2019-09-15:
Fixed with latest Seesaw Release.

closed 2019-09-15 by makermelissa