adafruit/circuitpython #2114: Feather nRF52840 Express Builtin NeoPixel Not Shared Correctly

jackdanielsmurphy, 2019-09-04:
I'm trying to use the builtin NeoPixel on the Feather nRF52840 Express. Here's my code:

```
pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)
while True:
    pixels[0] = (255, 0, 0)
    sleep(1)
```

The NeoPixel flashes between red and green. I assume this is because the semaphore system is interfering.
Looks like the builtin NeoPixel is not be shared correctly.

tannewt, 2019-09-04:
The supervisor may be setting the green color even though the neopixel should be used by the user's code.

siddacious, 2019-09-05:
I _may_ have seen this on one of my custom boards that has no on-board status led but the board defs are set to use a normally attached dotstar string. If I recall correctly when I `fill`'d the strip the first dotstar would remain a status indicator.

jerryneedell, 2019-09-09:
FYI -- at the REPL the NeoPixel is white

```import board
import neopixel
pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)
pixels[0] = (255, 0, 0)
```
turns it RED, but then it goes back to white within a second.

rdagger, 2019-09-09:
I confirmed the NeoPixel is basically unusable because the semaphore system is competing with the code for control of the NeoPixel.

jerryneedell, 2019-09-10:
Hmm, since these point to the same pin, it seems like supervisor “should” be well behaved..
https://github.com/adafruit/circuitpython/blob/master/ports/nrf/boards/feather_nrf52840_express/mpconfigboard.h#L37
https://github.com/adafruit/circuitpython/blob/master/ports/nrf/boards/feather_nrf52840_express/pins.c#L30

dhalbert, 2019-09-10:
`clear_temp_status() in `supervisor/shared/rgb_led_status.c` does not check for `neopixel_in_use`, unlike the other status neopixel writing routines in that file. I think maybe it should. I guess the question is whether the atmel-samd boards have the same behavior here. I looked at the uses of these routines and I don't see that the nrf port does something very different.

dhalbert, 2019-09-10:
OK, this is actually a manifestation of a bug in `atmel-samd/tick.c`, plus the bug in `clear_temp_status()`. It's supposed to do a filesystem flush once a second, but it's failing to do that. So the nrf port is actually doing the right thing, but in the process of checking whether to flush the filesystem, it's changing the status neopixel color without checking whether someone else is using neopixel.

The filesystem flushing should probably not fiddle with the status color unless it actually does a write. Right now it does it every time a flush is checked for, even if nothing happens.

PR #2136 (2019-09-10): fix atmel-samd filesystem_tick'ing; clear_temp_status() should check for status indicator in use
Fixes #2114.

- `clear_temp_status()` now checks whether status NeoPixel or DotStar is in use.
- in `atmel_samd/tick.c`, `SysTickHandler()` ws not calling `filesystem_tick()`

There is very similar code in the different versions of `ports/*/tick.c` which should be factored out eventually into common code. That's part of why the second bug above existed. Since we should rewrite the timing code anyway for #696 (sleep and wakeup), I don't think it's worth refactoring right now.
closed 2019-09-10 by tannewt