adafruit/circuitpython #2125: STM32F412 Discovery board lacks USB support

hierophect, 2019-09-06:
Should be updated once https://github.com/hathach/tinyusb/issues/103 is completed.

tannewt, 2019-09-09:
@hierophect This is done in TinyUSB. Please pull it in and get it going.

hierophect, 2019-09-09:
@tannewt yep got the notification this morning. I notice it isn't represented in Circuitpython master yet, despite it being on the develop branch and .gitmodules not tying it to a particular commit. Is that just a git thing, that it continues to be locked to a commit until a PR goes through? How does that update process work, exactly?

tannewt, 2019-09-09:
Git submodules are always tied to a particular commit. To update them, sync CircuitPython and then change into the submodule directory. Once there, fetch origin and checkout the remote branch (origin/develop). Now move back to a CircuitPython folder to do a test build. `git status` should show "new commits" for the submodule. You may need to `git add` it for it to be in the next commit but I'm not positive on that. Then, commit it like any normal commit and push to a branch to PR.

PR #2132 (2019-09-09): STM32: Add USB support to F412 Discovery 
This PR adds USB support to the F412 Discovery, reflecting the most recent port in tinyUSB. Includes revisions to the linker for filesystem support. Updates tinyUSB to latest develop release. 

Closes #2125.
closed 2019-09-09 by dhalbert