adafruit/circuitpython #2131: SAMD51 SPI Slave Mode

gingle, 2019-09-09:
I would like to use the Metro M4 as a remote controller run by a master microcontroller. I need to read A/D results and send D/A levels as well as set logic pins. I am worried that the I2C and RX/TX lines are not fast enough and perhaps not as free from errors.

tannewt, 2019-09-09:
@gingle It's not a high priority for us but we'd love a PR if you implement it yourself. We're also happy answering questions to get you started on it. I'd suggest looking at I2CSlave as a model for it.

gingle, 2019-09-09:
Can you point me to the I2CSlave code?

tannewt, 2019-09-09:
https://github.com/adafruit/circuitpython/tree/master/ports/atmel-samd/common-hal/i2cslave
https://github.com/adafruit/circuitpython/tree/master/shared-bindings/i2cslave

Want me to reopen? It's still ok to leave open for the long term.

gingle, 2019-09-09:
I would leave it open for the long term. I have a lot on my plate so my work on this will be intermittent. Thanks for the links.

tannewt, 2019-09-09:
Ok, reopened! No worries if it takes time. Please reach out if you have questions! Thanks!

wallarug, 2019-10-07:
@gingle  have you made any progress on this?  I'm very interested in getting this going.

gingle, 2019-10-07:
> 
> 
> @gingle have you made any progress on this? I'm very interested in getting this going.

Not for now. I have been sidetracked for awhile.

Meschmid23, 2019-11-07:
I also would like to use the SAMD51 (Adafruit feather M4) as a SPI slave. Please let us know once progress has been made.

Randall-Scharpf, 2024-05-10:
Hi all! Not sure what the current status of progress is on this, but for a set of boards I'm working on, we've needed to get SPI communication between two SAM D51 boards running CircuitPython.* Our code is a bit of a mess right now since it's just for our internal project, but (admittedly not having looked through the commits mentioned here) it might be helpful developing this feature, if there's anyone else still interested in getting it running. Namely, I've noticed some talk in the issue linked here about messing with the DMA controller to get it working, and we've had to do that. We've got a fork at https://github.com/Bruin-Spacecraft-Group/bruinspace-circuitpython/tree/cfa4425d3443ebb54ace97f78d477a34079b44cd if you'd like to take a look. If there's anything you'd like from us to help with establishing these changes for others or documenting what we've done, let us know!

jepler, 2024-05-10:
When an issue is tagged "long term", Adafruit isn't prioritizing work on it. But we're always happy if a community member can contribute something!

tannewt, 2024-05-13:
@Randall-Scharpf The easiest way for us to review what you've done is to make a pull request to this repo. That will show us a diff of changes that we can comment on. https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github/create-your-pull-request

closed 2026-05-16 by dhalbert