adafruit/circuitpython #2137: 5.0.0-alpha.2 displayio Label text doesn't update on 2-Color EPD FeatherWing

anecdata, 2019-09-10:
Environment:
Adafruit Feather M4 Express with samd51j19
5.0.0-alpha.2 on 2019-09-04
CP Lib Bundle 20190907

The code below attempts to update the text of a displayio Label on a 2-color EPD FeatherWing, but only the first iteration is displayed. Text does not update after the first iteration on the 2-color EPD. It goes through its flashing motions, but the same text appears again in every loop. print() output shows the code is progressing through its loops with the 2-color EPD.

Similar code works on PyPortal and on other FeatherWings, including 2.4" & 3.5" TFTs, OLED, and the 3-color EPD. 

However, inserting:
```
display.show(None)
display.refresh()
```
...into the loop allows the text of the label to get updated properly on the 2-color EPD during every loop. Seems like a boundary condition / element initialization issue that switching to the terminal does cleanly.

The 3-Color EPD will run the code with or without the switch to the terminal. Only the 2-Color EPD will _not_ run the code as shown below (with `display.show(None)` commented out).

I've sped up the `seconds_per_frame` in this example, but it behaves the same with the default 180 seconds.

This issue was moved from :https://github.com/adafruit/Adafruit_CircuitPython_SSD1675/issues/2 (which will be closed).

```
import os
import time
import board
import terminalio
import displayio
from adafruit_display_text.label import Label

# 2-Color EPD FeatherWing
from adafruit_ssd1675 import SSD1675

# 3-Color EPD FeatherWing
from adafruit_il0373 import IL0373

# Feather M4 Express + choose your own EPD adventure:
ssd1675 = True
il0373 = False

displayio.release_displays()
cs = board.D9
dc = board.D10

if ssd1675:
    display_bus = displayio.FourWire(board.SPI(), command=dc, chip_select=cs, baudrate=1000000)
    time.sleep(1)
    display = SSD1675(display_bus, width=250, height=122, rotation=90, seconds_per_frame=15)
elif il0373:
    display_bus = displayio.FourWire(board.SPI(), command=dc, chip_select=cs, baudrate=1000000)
    time.sleep(1)
    display = IL0373(display_bus, width=212, height=104, rotation=90, highlight_color=0xff0000, seconds_per_frame=30)

# displayio

SPLASHWIDTH = display.width
SPLASHHEIGHT = display.height
SPLASHMAX = 2
splash = displayio.Group(max_size=SPLASHMAX)

# Group 0
background_bitmap = displayio.Bitmap(SPLASHWIDTH, SPLASHHEIGHT, 1)
background_palette = displayio.Palette(1)
background_palette[0] = 0x0000
background = displayio.TileGrid(background_bitmap, pixel_shader=background_palette)
splash.append(background)

# Group 1
splash.append(displayio.Group(max_size=1, scale=1, x=0, y=0))  #dummy
font = terminalio.FONT
text_area = Label(font, text="01234567890123456789", color=0xFFFFFF)
text_area.x = 5
text_area.y = 25
splash[1] = text_area

# print the machine, OS, and bundle. VERSIONS.txt is included in all CP Library Bundles.
# you can safely comment out the next 3 lines
with open('/VERSIONS.txt') as f:
    cp_lib_bundle = f.readline().strip()
print(os.uname()[4], os.uname()[3], cp_lib_bundle)

loop = 0
while True:
    # Wait, then update and display Label

    time.sleep(display.time_to_refresh)
    try:
        text_area.text = str(loop) + ' ' + str(display.time_to_refresh)

        print("Text", loop, display.time_to_refresh)
        display.show(splash)
        display.refresh()
    except RuntimeError as e:  # Refresh too soon
        print(e)

    """ UNCOMMENT THE FOLLOWING BLOCK TO MAKE IT WORK ON 2-Color EPD FeatherWing
    # Wait, then display default CircuitPython terminal

    time.sleep(display.time_to_refresh)
    try:
        print("None", loop, display.time_to_refresh)
        display.show(None)
        display.refresh()
    except RuntimeError as e:  # Refresh too soon
        print(e)
    """

    loop += 1
```

anecdata, 2019-09-30:
This seems to have been fixed somewhere along the way. Working now in alpha.4 with latest libraries. I hadn't tested again until the past couple of days since I hadn't seen anything in release notes that looked relevant. Closing this issue.

anecdata, 2019-09-30:
Sorry, the example still doesn't work. But my more complex app now does. I'll look into the usage differences to see what the magic is.

anecdata, 2019-09-30:
Well, the usage difference is that I'd completely forgotten that I employ the usual trick of re-creating the object each time when some update functionality isn't working (duh).

But, played with changing x and y in the loop and some perhaps helpful behaviors become apparent. With the last part of the code changed to:
```
# Group 1
splash.append(displayio.Group(max_size=1, scale=1, x=0, y=0))  #dummy
font = terminalio.FONT
text_area = Label(font, text="01234567890123456789", color=0xFFFFFF)
x = 5
y = 25
text_area.x = x
text_area.y = y
splash[1] = text_area

loop = 0
while True:
    # Wait, then update and display Label

    time.sleep(display.time_to_refresh)
    try:
        x += 1
        y += 1
        text_area.x = x
        text_area.y = y
        text_area.text = str(loop) + '~' + str(display.time_to_refresh) + '~' + str(loop)

        print("Text", loop, display.time_to_refresh)
        display.show(splash)
        display.refresh()
    except RuntimeError as e:  # Refresh too soon
        print(e)

    # time.sleep(60)

    """ UNCOMMENT THE FOLLOWING BLOCK TO MAKE IT WORK ON 2-Color EPD FeatherWing
    # Wait, then display default CircuitPython terminal

    time.sleep(display.time_to_refresh)
    try:
        print("None", loop, display.time_to_refresh)
        display.show(None)
        display.refresh()
    except RuntimeError as e:  # Refresh too soon
        print(e)
    """

    loop += 1
```
(again, you can change the seconds_per_frame KW arg when you create the display to speed up iterations)

With this code, you can see a similar behavior to #2138 where the x-axis goes to zero, but also that _some_ of the text now updates, and only some of that updates only on odd iterations. Seems like some incomplete init, and perhaps a bug with content change detection (although I didn't think EPDs did partial update).

dhalbert, 2019-10-31:
Hi @anecdata, could you retest on 5.0.0-alpha.4? Thanks!

anecdata, 2019-10-31:
5.0.0-alpha.4 on 2019-09-15 with CP Lib Bundle 20191029
doesn't appear to improve or change the behavior of either example.

tannewt, 2019-11-26:
I suspect this is the same issue that #2330 fixes.

anecdata, 2019-11-26:
Latest S3 seems to be that merge, but example behavior is the same with:
Adafruit CircuitPython 5.0.0-beta.0-67-gf7426e0e6 on 2019-11-26; Adafruit Feather M4 Express with samd51j19
+
adafruit-circuitpython-bundle-5.x-mpy-20191126

anecdata, 2021-02-09:
Tested the 2-color EPD FeatherWing again with 6.2.0-beta.1 on a FeatherS2 with adafruit-circuitpython-bundle-6.x-mpy-20210209. Issue still occurs, changes to label text aren't shown with a simple `display.refresh()`. This sequence gets it to work:
```
display.show(None)
display.show(splash)
display.refresh()
```
Neither the 3-color EPD FeatherWing nor the MagTag exhibit this issue.

dhalbert, 2021-06-15:
@anecdata Does this problem still exist?

anecdata, 2021-06-16:
Re-tested both code examples above, and the issue is still present in
`Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; Adafruit Feather M4 Express with samd51j19`

tannewt, 2023-04-26:
I think this is fixed by https://github.com/adafruit/Adafruit_CircuitPython_SSD1675/pull/16

anecdata, 2023-04-26:
Boom. Fixed by https://github.com/adafruit/Adafruit_CircuitPython_SSD1675/pull/16. Thanks @tannewt! Closing.

bradanlane, 2023-11-19:
I see this issue has been closed.
I am testing an SSD1608 and see that the `text_area.y` is ignored. i.e. on first refresh the test is in the correct location. On the subsequent refresh the x position is correct the y position it at the top of the display.

The workaround is to use the same trick suggested by @anecdata ...

CODE:
```
displayio.release_displays()
spi = busio.SPI(SCK_PIN, MOSI_PIN)
display_bus = displayio.FourWire(spi, command=DC_PIN, chip_select=CS_PIN, reset=RESET_PIN, baudrate=1000000)
time.sleep(1)
display = adafruit_ssd1608.SSD1608(display_bus, width=200, height=200, busy_pin=BUSY_PIN)

display_group = displayio.Group()
background_bitmap = displayio.Bitmap(INK_WIDTH, INK_HEIGHT, 1)
palette = displayio.Palette(1)
palette[0] = INK_WHITE

group_tile = displayio.TileGrid(background_bitmap, pixel_shader=palette)
display_group.append(group_tile)
display.root_group = display_group

display.show(display_group)
display.refresh()
time.sleep(display.time_to_refresh + 0.1)

display_group.append(displayio.Group(scale=4, x=70, y=100))
text_area = label.Label(terminalio.FONT, text="LED", color=INK_BLACK)

text_area.x = 70
text_area.y = 100
text_area.scale = 4
display_group[1].y=100
display_group[1] = text_area

# without these two lines, the "LED" text will appear at the top of the display rather than at y=100
display.show(None)
display.show(display_group)

display.refresh()
```

_I can open a new issue, if that is the correct nedt step._