adafruit/circuitpython #2138: 5.0.0-alpha.2 displayio Group loses x-coordinate on 2-Color EPD FeatherWing

anecdata, 2019-09-10:
Environment:
Adafruit Feather M4 Express with samd51j19
5.0.0-alpha.2 on 2019-09-04
CP Lib Bundle 20190907

The code below attempts to implement the Sprite Sheet example from:
[https://learn.adafruit.com/circuitpython-display-support-using-displayio/sprite-sheet](https://learn.adafruit.com/circuitpython-display-support-using-displayio/sprite-sheet)
...by adapting it to the timing requirements of EPD displays (e.g., `time.sleep(display.time_to_refresh)`)

The first iteration through the main loop will execute properly, but on subsequent iterations, the sprites will be displayed at x = 0 on the 2-color EPD FeatherWing.

You can shorten the iterations from 180 seconds by adding the `seconds_per_frame` KW param when you create the display. 10 seconds is safe for the 2-color (it will go as low as <2 seconds); 30 seconds is safe for the 3-color.

The sprites remain in the proper location on the 3-color EPD FeatherWing with the code below (swap the True/False on the board flags near the top). But sometimes the display on the 3-color FeatherWing won't refresh... that will be the subject of a separate issue though I suspect similar init/deinit causes on all of these issues.

Similar code (directly from the Learn guide) works as expected on PyPortal and on other FeatherWings, including 2.4" & 3.5" TFTs, and OLED.

```
import os
import time
import board
import terminalio
import displayio
import adafruit_imageload
from adafruit_display_text.label import Label

# 2-Color EPD FeatherWing
from adafruit_ssd1675 import SSD1675

# 3-Color EPD FeatherWing
from adafruit_il0373 import IL0373

# Feather M4 Express + choose your own EPD adventure:
ssd1675 = False
il0373 = True

displayio.release_displays()
cs = board.D9
dc = board.D10

if ssd1675:
    display_bus = displayio.FourWire(board.SPI(), command=dc, chip_select=cs, baudrate=1000000)
    time.sleep(1)
    display = SSD1675(display_bus, width=250, height=122, rotation=90)
elif il0373:
    display_bus = displayio.FourWire(board.SPI(), command=dc, chip_select=cs, baudrate=1000000)
    time.sleep(1)
    display = IL0373(display_bus, width=212, height=104, rotation=90, highlight_color=0xff0000)

# print the machine, OS, and bundle. VERSIONS.txt is included in all CP Library Bundles.
# you can safely comment out the next 3 lines
with open('/VERSIONS.txt') as f:
    cp_lib_bundle = f.readline().strip()
print(os.uname()[4], os.uname()[3], cp_lib_bundle)

# the following code has been adapted for EPD from:
# https://learn.adafruit.com/circuitpython-display-support-using-displayio/sprite-sheet
# the BMP file, "/cp_sprite_sheet.bmp", is available there

# Load the sprite sheet (bitmap)
sprite_sheet, palette = adafruit_imageload.load("/cp_sprite_sheet.bmp",
                                                bitmap=displayio.Bitmap,
                                                palette=displayio.Palette)

# Create a sprite (tilegrid)
sprite = displayio.TileGrid(sprite_sheet, pixel_shader=palette,
                            width = 1,
                            height = 1,
                            tile_width = 16,
                            tile_height = 16)

# Create a Group to hold the sprite
group = displayio.Group(scale=1)

# Add the sprite to the Group
group.append(sprite)

# Add the Group to the Display
display.show(group)

# Set sprite location
group.x = 60  # was 120
group.y = 20  # was 80

# Loop through each sprite in the sprite sheet
source_index = 0
while True:
    sprite[0] = source_index % 6
    source_index += 1

    time.sleep(display.time_to_refresh)
    try:
        print(source_index, display.time_to_refresh)
        display.refresh()
    except RuntimeError as e:  # Refresh too soon
        print(source_index, display.time_to_refresh, e)
```

dhalbert, 2019-10-31:
Hi @anecdata, could you retest on 5.0.0-alpha.4? Thanks!

anecdata, 2019-10-31:
5.0.0-alpha.4 on 2019-09-15 with CP Lib Bundle 20191029
doesn't appear to improve or change the behavior of either example.

(Note that the code as shown above has the 3-color EPD enabled, just swap the True / False in lines 17 & 18.)

tannewt, 2019-11-26:
I suspect this is the same issue that #2330 fixes.

anecdata, 2019-11-26:
Latest S3 seems to be that merge, but example behavior is the same with:
Adafruit CircuitPython 5.0.0-beta.0-67-gf7426e0e6 on 2019-11-26; Adafruit Feather M4 Express with samd51j19
+
adafruit-circuitpython-bundle-5.x-mpy-20191126

dhalbert, 2021-06-15:
@anecdata Does this problem still exist?

tannewt, 2023-04-26:
I think this is fixed by https://github.com/adafruit/Adafruit_CircuitPython_SSD1675/pull/16

anecdata, 2023-04-27:
Code still works on PyPortal (when fixed up for built-in display and no `time_to_refresh`), but I can't get any images to display on the 2-color EPD FeatherWing for some reason, not imageload and not OnDiskBitmap. With or without the little endian library change. I'll keep poking at it. Maybe some API changed that I didn't keep up with.

tannewt, 2023-04-27:
Darn. Ok. Thanks for testing!