adafruit/circuitpython #2140: DisplayIO: SH1106 & SSH1106 OLED display drivers

NightSkySK, 2019-09-10:
Could you please consider to add support for OLED displays I2C with [SSH1106](https://www.amazon.com/Ponis-Limos-Display-Interface-Communicate-SSH1106/dp/B07WG6FPBP)
and [SH1106](https://www.amazon.com/Serial-128X64-Display-Module-Arduino/dp/B07QY5DRVB)

tannewt, 2019-09-10:
Related forum thread: https://forums.adafruit.com/viewtopic.php?f=60&t=141184

LintangWisesa, 2020-05-08:
Anyway, you can use ```Luma.OLED``` display library for most OLED displays. It can be used for Interfacing OLED matrix displays with the SSD1306, SSD1309, SSD1322, SSD1325, SSD1327, SSD1331, SSD1351, SSD1362 or SH1106 driver in Python 2 or 3 using I2C/SPI on the Raspberry Pi and other linux-based single-board computers. More info: [Luma.OLED docs](https://luma-oled.readthedocs.io/en/latest/index.html).

tannewt, 2020-05-08:
@LintangWisesa This issue is for CircuitPython which runs on microcontrollers rather than single board computers. Please keep the discussion specific to CircuitPython.

ladyada, 2021-07-21:
https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SH1106

closed 2021-07-21 by ladyada