adafruit/circuitpython #2141: Add board support for SAMD51 Thing Plus

jepler, 2019-09-10:
The SAMD51 Thing Plus from sparkfun looks like a good candidate for CircuitPython. Add support for this board, or determine why it's not feasible.

https://learn.sparkfun.com/tutorials/samd51-thing-plus-hookup-guide/all

Note that while this isn't called out clearly in feature lists, it features a 4Mb (256KB) SPI flash, verified by jasonp on discord

jepler, 2019-09-10:
I have one of these on order, and hope to knock it out some even next week.  If it's been longer than that, someone else should feel free to pick this up.

ladyada, 2019-09-10:
hiya @santaimpersonator please try this out and submit a PR so we can generate builds automagically

santaimpersonator, 2019-09-13:
@jepler @ladyada I'll try to get to this on Monday/Tuesday. If not, I'll be on vacation for the next few weeks and will start on it when I get back. I have added this to my "to do" list (*\*just an FYI so it doesn't seem like I am neglecting this*). Thanks!

jepler, 2019-09-13:
@santaimpersonator I will let you take the lead on this, thank you!

santaimpersonator, 2019-10-01:
So, we have a [fork of the CircuitPython repository](https://github.com/sparkfun/circuitpython). I think I made the correct changes... @jepler @ladyada could you doublecheck that I got everything correct before I submit a pull request.

[Flash Memory Configuration:](https://github.com/sparkfun/circuitpython/blob/master/supervisor/shared/external_flash/devices.h#L88)
I think I configured the flash chip properly. The datasheet is linked in the comments.
* I couldn't find a listed startup time, so copied the value from the AT25DF081A (`.start_up_time_us = 10000, \`).
* I found the bitmask for the Quad Enable bit, but I wasn't sure if I set it up correctly (`.quad_enable_bit_mask = 0x01, \`).

[Bootloader:](https://github.com/sparkfun/circuitpython/blob/master/ports/atmel-samd/boards/sparkfun_samd51_thing_plus/mpconfigboard.mk#L1)
* I think I selected the proper ld file (`LD_FILE = boards/samd51x20-bootloader-external-flash.ld`).

[Board Configuration:](https://github.com/sparkfun/circuitpython/blob/master/ports/atmel-samd/boards/sparkfun_samd51_thing_plus/mpconfigboard.h)
I wasn't sure if I configured some of the settings properly:
* Line 16: MICROPY_PORT_A (I specified the SPI Flash pins... copied from Metro M4)
* Line 21: Auto-reset delay (wasn't sure if this was necessary... copied from Metro M4)
* Line 25 & 27: Memory sizing (comments said to leave alone... copied from Metro M4)

[Pins:](https://github.com/sparkfun/circuitpython/blob/master/ports/atmel-samd/boards/sparkfun_samd51_thing_plus/pins.c)
I was looking at the Metro M4... what is the difference between MP_OBJ_NEW_QSTR vs MP_ROM_QSTR?

santaimpersonator, 2019-10-01:
I was able to build the `firmware.uf2` file (linked below) and it appears to work. However, I am having issues testing if the flash chip is working properly.

I was following the [Adafruit guide for the Feather M0](https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/circuitpython-storage). After my attempt to use the Arduino Example to read the flash failed, I was trying to upload another example. I deleted the `boot.py` and `code.py` files; then I acidentally reset the board. Now I can't do anything CircuitPython related. I can upload the firmware again, but the USB mass storage drive doesn't show up anymore; only when I force it into the bootloader (to upload the `firmware.uf2` file). I can still load Arduino examples, but nothing is accessible for CircuitPython. I am pretty sure I just need to reflash the microcontroller again.

[SparkFun SAMD 51 Firmware.zip](https://github.com/adafruit/circuitpython/files/3678728/SparkFun.SAMD.51.Firmware.zip)

ladyada, 2019-10-02:
hiya sounds like the flash chip setup may not be right. you can start a PR if you'd like. is @edspark able to help out? they did the other boards. let us know either way!

ladyada, 2019-10-02:
we also have a guide here you can follow https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

tannewt, 2019-10-02:
Here is a diff for folks who want to see the changes: https://github.com/adafruit/circuitpython/compare/master...sparkfun:master

Flash start up time should be ok. For the bitmask you'll want 0x02 since quad enable is bit 1 in the status register. You could try lowering the maximum speed to ensure it's not related to that.

Linker script looks right.

For board config, you'll want to add the QSPI pins if you switch to that. In SPI mode, it should not reset the pins automatically. I don't think we added that for QSPI though.

I didn't realize there was a difference between MP_ROM_QSTR and MP_OBJ_NEW_QSTR. It looks like it only matters on systems with 64 bit words (unix) which is why we're inconsistent about it. I'd suggest MP_ROM_QSTR for clarity.

santaimpersonator, 2019-10-03:
As a follow-up, I realized that the [tutorial I was following](https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/circuitpython-storage) was for storing information to the SAMD51 memory and not the the flash chip.

Digression:
However, I tried to erase and re-flash the original firmware for the SAMD51 (itself) and the same issue still persisted. I can see the board as a mass storage device and upload Arduino code to it. However, after I upload the CircuitPython firmware, the board won't show up as a mass storage device named `CIRCUITPYTHON` as it did originally. I can still force the board into the bootloader and use it with Arduino, but it is bricked as far as anything Circuit Python is related (not sure what is going on there).

santaimpersonator, 2019-10-03:
@tannewt 
* Good catch... I guess my brain is still on vacation. I probably should update the quad SPI setting for the flash chip to `.supports_qspi = true, \` as well.
* I'll try lowering the speed; however, I thought these settings were specific for the chip and that was going to be limited by the baud rate setting in the [mpconfigboard header file](https://github.com/adafruit/circuitpython/compare/master...sparkfun:master#diff-4aa224cd8c52b041992247209c614ef3R7) anyways.
* Thanks, that seemed like the best fit, but I wasn't perfectly sure.
* I thought QSPI was for "quad" SPI. I don't think it is implemented with the flash chip on the SAMD51 Thing Plus([schematic](https://cdn.sparkfun.com/assets/8/2/1/d/f/SAMD51_Thing_Plus_v10.pdf)).
  * I set the data lines in Line 16: MICROPY_PORT_A mostly because it was set for the Metro M4. Not really sure if it was needed.
* Thanks, I was kind of just guessing with that. (Most of it was beyond my understanding.)

santaimpersonator, 2019-10-03:
@ladyada 
* Thanks... I'll double check it with @tannewt's suggestions.
* @edspark helped walk me through your process, but didn't really have an answer for those specific questions.
* Is [this Arduino example](https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/using-spi-flash) the only way to test the flash chip through CircuitPython?
  * I finally got the **fatfs_circuitpython example** to compile with our board (we don't use QSPI), but I am getting an `Error, failed to initialize flash chip!`... which I think is from the wrong flash chip is being associated with the SAMD51 on our board. However, I am not sure where the flash chip is defined for the boards in the library (I saw a [flash_devices.h](https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/flash_devices.h) header file, but wasn't sure how it was accessed).

ladyada, 2019-10-03:
you're really close!
it looks like the flash memory wired as plain-SPI not QSPI, that's OK, you can use the SPI Flash setup like from the feather m0 express
https://github.com/adafruit/circuitpython/tree/master/ports/atmel-samd/boards/feather_m0_express

SPIFlash arduino library should work, if you have to have the SPI peripheral set up for use, which depends on your board definition. but if you pick the right SPI port and the right CS pin, then you should be able to read the JEDEC ID, at lesat

tannewt, 2019-10-03:
Note that if you format the filesystem with Arduino, it may not set the fslabel to CIRCUITPY like the formatter in CircuitPython will. It should still show up though I think.

One thing to try is from the REPL do:
```
import os
os.listdir()
```
That will tell you if CircuitPython sees the flash but the USB isn't happy.

You can also try formatting the flash from CircuitPython with these instructions: https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#easiest-way-use-storage-dot-erase-filesystem-20-23

ladyada, 2019-11-15:
@santaimpersonator hihi is there any news on this issue? we'd love to add it!

ladyada, 2019-12-20:
@santaimpersonator hihi checking again - will you be able to create the PR? please let us know either way!

santaimpersonator, 2019-12-20:
Sorry, I haven't forgotten about this. I actually just freed up from my last project, so now I have time dive into this again. I am still looking into the same issue (to make sure that the flash memory is working).

AndyEngland521, 2019-12-31:
Looks like an issue with our board definitions prevented the user from accessing external flash, I've fixed this in the latest release (1.7.4) and am able to get a proper JEDEC ID from the flash chip using the SdFat_circuitpython example. However, it seems like we are unable to mount the filesystem

ladyada, 2020-02-25:
added and merged - thanks @jepler :)

closed 2020-02-25 by ladyada