adafruit/circuitpython #2142: Reduce overhead of background tasks

dhalbert, 2019-09-10:
Opening new issue to narrow focus of #1933. See that issue for original details.

`run_background_tasks()` calls a number of routines that need to run in the background. Some of these do extensive checking to see if there's anything to do. The overhead of these routines could be reduced to save a substantial amount of time (like another factor of 2).

`run_background_tasks()` is called in after the completion of every jump opcode and every return in `vm.c`. So an improvement here would speed up the VM substantially. It is also called in busy-wait loops (as `RUN_BACKGROUND_TASKS`).

Some other background tasks are done at intervals determined in `SysTickHandler()` in `ticks.c`

As an example, we run the TinyUSB background task on every call, but for full-speed USB, we need only run this every ms (can consult with @hathach about this). Similarly, other tasks need only be done at certain intervals. The intervals for needing to run tasks could be triggered in the SysTickHandler in ticks.c, and the checks could be simple boolean checks. @tannewt can give guidance on how to check for displayio refresh tasks.

jepler, 2019-09-10:
Could we start by saying that if there hasn't been a 1ms tick, we won't even check whether any background task needs to be done?  

Any advice on how to benchmark this and measure concrete speed-up?  pystone benchmark is one I've used in the past when estimating how much nRF lto increased speed.

dhalbert, 2019-09-10:
@tannewt thought that 1ms ticks might produce frame rates that were somewhat off for higher-frame rate display refreshes, so it'd be good to discuss that with him.

In #1933, I just measured a very simple loop to get what the maximum raw speedup was. I think that might be sufficient here: we're not targeting a particular speed, but just looking to minimize the background overhead.

tannewt, 2019-09-11:
The frame rates are currently off already so I wouldn't worry about it. I was getting ahead of myself when I pointed that out.

dhalbert, 2019-10-31:
Ideally, we would do this by 5.0.0 GA.

closed 2019-12-04 by jepler

ladyada, 2019-12-04:
amazing teamwork everyone !