adafruit/circuitpython #2145: Meowbit support?

Mr-Coxall, 2019-09-11:
Can the Meowbit be added to have CircuitPython support?
https://www.kittenbot.cc/collections/frontpage/products/meowbit-codable-console-for-microsoft-makecode-arcade

tannewt, 2019-09-11:
Looks like it is an STM32F401. We're currently targeting the F411, F412 and F405 but may be able to support the F401 in the longer term.

hierophect, 2019-09-11:
The F401 is very similar to the F411. It may be trivially simple to add. However, since I don't have a viable dev board for it on hand, I think this is probably best scheduled after we have modules like displayio up and running.

ladyada, 2019-09-15:
@hierophect feel free to pick up a https://www.adafruit.com/product/4324 on your travels :)

PR #2404 (2020-01-30): STM32: Meowbit support
This PR adds support for the Meowbit board by Kittenbot. Adds definitions for the STM32F401RE, a new MCU for the port. Requires #2395 and #2379. Resolves #2196, resolves #2145.

Bugs fixed as a part of debugging this board:
 - A dangling pointer error was present in PWMOut, which caused erratic issues when using PWMOut and the backlight in DisplayIO . Resolves #2524, resolves #2444. 
 - Resolves a vector table relocation issue that would be present for any UF2 bootloader on STM32
 - Fixes a pin claiming issue that would occur in rare cases and prevent pins from being re- [... cut]
closed 2020-01-30 by tannewt