adafruit/circuitpython #2152: SD Card incompatibility from #245 not resolved?

AdinAck, 2019-09-13:
My SD Card had the same error from issue #245 so I downloaded the fixed version that was made, that they said was merged in but it looks like it wasn't because downloading the script from that thread fixed the problem.

tannewt, 2019-09-13:
@AdinAck please post a picture of your SD card. Many of them are just a little different and need to be fixed separately.

AdinAck, 2019-09-13:
It’s exactly the same as the ones in the fix (Samsung EVO 32Gb). I solved
the error by downloading the “adafruit_sdcard.py” from
https://github.com/adafruit/Adafruit_CircuitPython_SD
and placing adafruit_sdcard.py next to code.py and importing it, instead of
adafruit_sdcard.mpy from the lib folder. Wouldn’t this mean the fixed code
is not in the latest library release? If that doesn’t make sense let me
know and I’ll send you a picture of the sd card.

Thanks!

tannewt, 2019-09-13:
Ah, ok. Yup, that may mean we've not released it. I'll do that now. Thanks!

tannewt, 2019-09-13:
Ok, released as 3.2.4. Will be in tomorrow's bundle. Thanks!

closed 2019-09-13 by tannewt