adafruit/circuitpython #2157: How to import large complicated python scripts from sd card?

AdinAck, 2019-09-15:
I have a grand central, and have successfully set up importing singular .py scripts from the sd card and am trying to get numpy working, but it is not working. The current error is:
AttributeError: 'module' object has no attribute 'array'
Which basically means it has not imported numpy properly XD.
Numpy is a series of folders instead of a single .py which I suspect is the issue. I was wondering if there is a way to make this work.
Thanks!

ladyada, 2019-09-15:
hiya - numpy is not going to be supported, its extremely CPython dependant!

closed 2019-09-15 by ladyada

tannewt, 2019-09-16:
@AdinAck Did you modify sys.path to include the folder on the sd card?

As @ladyada says it's highly likely that numpy will take too much memory and/or require native modules we don't support.

AdinAck, 2019-09-16:
Yes the sys.path included the SD card. I have given up on using numpy. Thanks for the help though~

Adin Ackerman

From: Scott Shawcroft
Sent: Sunday, September 15, 2019 8:51 PM
To: adafruit/circuitpython
Cc: AdinAck; Mention
Subject: Re: [adafruit/circuitpython] How to import large complicated pythonscripts from sd card? (#2157)

@AdinAck Did you modify sys.path to include the folder on the sd card?
As @ladyada says it's highly likely that numpy will take too much memory and/or require native modules we don't support.

dhalbert, 2019-11-04:
@AdinAck We recently heard about this interesting library: https://micropython-ulab.readthedocs.io/en/latest/, which might be able to fit on certain CircuitPython builds.

AdinAck, 2019-11-04:
That’s awesome I’ll check it out!