adafruit/circuitpython #2172: STM32: Improve I2C reset workaround

hierophect, 2019-09-20:
I2C currently has problems resetting after a soft reboot, such as after a write to disk. The error is actually not listed in the F4 errata, but follows the exact pattern of the "I2C analog filter may provide wrong value" error listed in the F10xx erratas [here](https://www.st.com/content/ccc/resource/technical/document/errata_sheet/f5/50/c9/46/56/db/4a/f6/CD00197763.pdf/files/CD00197763.pdf/jcr:content/translations/en.CD00197763.pdf) (2.14.7) and [here](https://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf#page26)(2.13.7). Stack exchange discussion is [here](https://electronics.stackexchange.com/questions/272427/stm32-busy-flag-is-set-after-i2c-initialization).

The current workaround for this error is crude and not fully implemented as advised in the errata, and should be improved to match the recommended solution.

tannewt, 2019-09-20:
[image]

hierophect, 2020-01-02:
Making a note of this forum issue (experiences internal displayio error on reset) to try and validate whether it's related: https://forums.adafruit.com/viewtopic.php?f=60&t=159872&p=789428#p789428

PR #2463 (2020-01-14): STM32: I2C fix & general busio cleanup
This PR revisits some of the differences that appeared between the busio modules as they were implemented and brings them closer together in terms of style and readability. Also reorganizes the fix for an issue with I2C that would occur on soft reboot - the code is now closer to comparable APIs like Arduino and MBED. Resolves #2172. Changed modules tested on STM32F405 feather and STM32F412 discovery.
closed 2020-01-14 by hierophect