adafruit/circuitpython #2174: SD Card Breakout + for Raspberry Pi

ankonaskiff17, 2019-09-21:
Could the SD Breakout + tutorial be updated to reflect wiring to Raspi to include schematic and appropriate code for CircuitPython for SPI and I2C

ladyada, 2019-09-21:
hiya you shouldnt be using an SD card breakout for raspberry pi, use a USB to SD card adapter instead. where is this guide? we'll add a note

ankonaskiff17, 2019-09-21:
It's in the LEARN link for the breakout if one is to purchase it if I recall. Not near my computer. I opted for this to log weather sensor data on loss of internet connectivity and/or utility power. Im already using PowerBoost1000c and INA219 as UPS setup. Will take a look at what you are suggesting.

Wayne Wells

ladyada, 2019-09-21:
when you are at your computer please give us the exact link you're referring to
thanx!

ankonaskiff17, 2019-09-21:
Here it is. Not so much that it says you can use with a Pi, as much as it doesn’t say you cant. As a newbie I saw the CircuitPython comments and assumed it would work,

 

http://learn.adafruit.com/adafruit-micro-sd-breakout-board-card-tutorial

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

ankonaskiff17, 2019-09-21:
Also, do you have a part number for item you say  should use? Can’t find and needs to be wired in to a proto board.

 

https://www.avast.com/antivirus

ankonaskiff17, 2019-09-21:
Think I overthought that deal. Just put a usb flash drive in open usb port should cover it??

 

https://www.avast.com/antivirus

ladyada, 2019-09-21:
yep. closing because this is not an issue with circuitpython

closed 2019-09-21 by ladyada

ladyada, 2019-09-22:
we added a page https://learn.adafruit.com/adafruit-micro-sd-breakout-board-card-tutorial/python-linux

ankonaskiff17, 2019-09-22:
Thank you. With the proliferation of microprocessors and single board computers it can be quite a challenge to sort out what does and doesn’t work with each other. 

 

https://www.avast.com/antivirus