adafruit/circuitpython #2187: PDMIn : The SPI pins combination used on the Monster M4SK is not supported in  5.0.0-alpha.4

Anton-2, 2019-10-02:
```
Adafruit CircuitPython 5.0.0-alpha.4 on 2019-09-15; Adafruit Monster M4SK with samd51j19
>>> import audiobusio
>>> import board
>>> mic = audiobusio.PDMIn(board.MICROPHONE_CLOCK, board.MICROPHONE_DATA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid clock pin
```

This is in  [common-hal/audiobusio/PDMIn.c](https://github.com/adafruit/circuitpython/blob/22b7050c4cd8f2cad31657f91dd4a217b0444a76/ports/atmel-samd/common-hal/audiobusio/PDMIn.c#L77), the hardcoded pin combinations are not ready for the Monster M4SK.

ladyada, 2019-10-02:
yeah :( we could not use the I2S pins because there's no I2S ont he G19 - so instead we connected them to an SPI port - PDM is SPI-like!
now...in theory, we could adapt PDMIn.c to allow using SPI instead! or maybe do a user-space library....
for now we're only supporting PDM mic within arduino :)

dhalbert, 2019-10-07:
I added a note explaining that I2S is not available to the Pinouts section in the Learn Guide for the board.

closed 2019-10-07 by dhalbert