adafruit/circuitpython #2188: Customize exactly what is frozen for frozen module

dhalbert, 2019-10-03:
We need to be able to choose which files to freeze and which to skip for a given frozen module. First use case: A `bluefruit.py` will be added to the CPX library, but there's no reason to freeze it on the CPX, since that board doesn't have BLE capability.

Originally I thought we might notate in a particular file whether it should be frozen or not for a particular board. However, it's probably easier to include some value in `mpconfiboard.mk` that specifies which files to skip (or include) for a particular frozen library.

dhalbert, 2019-10-03:
Strawman example:
```
# Include these Python libraries in firmware.
FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_BusDevice
FROZEN_MPY_DIRS += $(TOP)/frozen/Adafruit_CircuitPython_CircuitPlayground
FROZEN_MPY_DIRS_EXCLUDE += $(TOP)/frozen/Adafruit_CircuitPython_CircuitPlayground/adafruit_circuitplayground/bluefruit.py
...
```

tannewt, 2019-10-03:
Why don't we make a separate library instead? I don't like the possibility of having to explain that only part of a library is available.

ladyada, 2019-10-03:
i'll vote for separate library!

kattni, 2019-10-03:
I don't understand why we'd ever be explaining that only part of the library is available - the things added to the `bluefruit` module will not work on the CPX anyway. None of the current example code will change. I wanted to keep it together because we designed this library to be expandable and I want to use it that way. The naming convention makes sense. I feel a separate library could prove to be confusing if/when someone tries to use this library on the CPB since it is named Adafruit_CircuitPython_CircuitPlayground. We'll be explaining something to someone either way. I was under the impression that Scott understood where I was coming from with this per a discussion on Discord, but I am apparently mistaken. I thought the plan was to move forward with this, and put an import exception into the beginning of the `express` module to handle attempting to import `bluefruit` on a CPX in a friendly manner.

tannewt, 2019-10-03:
I worry that having a general mechanism to do it will lead to us using it for other libraries.

Do we have the bluefruit library sketched out yet? Do we know it won't fit in the CPX?

dhalbert, 2019-10-03:
I was thinking about a library kind of like the Crickit library, not necessarily Circuit-Playground dependent, so yes, it could be separate. But a CPB-dependent library could know about the accelerometer, etc. (but maybe it shouldn't).

This is probably not the place to discuss the library. I opened this issue on a mistaken assumption about a decision, so I think I'll close it for now.

closed 2019-10-03 by dhalbert

ladyada, 2019-10-03:
all good - lets chat on monday!

kattni, 2019-10-03:
I want it to be a Bluefruit library like the CPX lib but for CPB where it makes getting started with CPB super simple in the exact same way CPX lib does for CPX. I want that experience for the CPB. It's only sketched out in my head so far.

ladyada, 2019-10-03:
ok! i think we're all in agreement - we'll work together to figure out what the library will be!