adafruit/circuitpython #2196: STM32: Support new dev boards

hierophect, 2019-10-03:
To be added:

- PYB LR Nano V2 [F411]
- F411 "Blackpill", generic [F411]
- Meowbit [F401]

ladyada, 2019-11-19:
matek betaflight controllers use stm32f405
https://www.amazon.com/MATEKSYS-F405-STD-Flight-Controller-STM32F405/dp/B079CN4QX3

hierophect, 2019-11-21:
@ladyada I'll need some new VIDs and PIDs for these boards, correct?

ladyada, 2019-11-22:
PYB Nano v2       VID 0x239A    PID 0x0067  # bootloader
                                PID 0x8067  # arduino
                                PID 0x8068  # circuitpython

F411 BlackPill    VID 0x239A    PID 0x0069  # bootloader
                                PID 0x8069  # arduino
                                PID 0x806A  # circuitpython

hierophect, 2019-11-22:
@ladyada thanks!

PR #2404 (2020-01-30): STM32: Meowbit support
This PR adds support for the Meowbit board by Kittenbot. Adds definitions for the STM32F401RE, a new MCU for the port. Requires #2395 and #2379. Resolves #2196, resolves #2145.

Bugs fixed as a part of debugging this board:
 - A dangling pointer error was present in PWMOut, which caused erratic issues when using PWMOut and the backlight in DisplayIO . Resolves #2524, resolves #2444. 
 - Resolves a vector table relocation issue that would be present for any UF2 bootloader on STM32
 - Fixes a pin claiming issue that would occur in rare cases and prevent pins from being re- [... cut]
closed 2020-01-30 by tannewt