adafruit/circuitpython #2199: `.Trashes` should be a directory, not a file

dhalbert, 2019-10-04:
`.Trashes`, created when a filesystem is initialized, should be a directory, not a file. See https://github.com/hathach/tinyusb/issues/182#issuecomment-538320398.

dhalbert, 2019-10-04:
OK, further research. We deliberately made `.Trashes` a file, not a directory, because for MacOS, that prevents a "trash can" from being created on the filesystem with deleted files. It works to do that on MacOS, but apparently iOS is not acting properly. So we could report this to Apple.

CollinCunningham, 2019-10-08:
@dhalbert Do you know if creating `.Trashes` as a file instead of a directory is an official solution from Apple or a hack/workaround?

dhalbert, 2019-10-08:
@CollinCunningham The technique appears multiple places, but nothing official that I can find. It does not appear to be documented, but then, Apple doesn't document a lot of stuff. I did some websearches on site:apple.com and could only find mentions in discussions.apple.com

dhalbert, 2019-10-31:
Did someone report this to Apple? I thought that was in the works.

CollinCunningham, 2019-11-01:
@dhalbert i believe pt filed a bug report

dhalbert, 2020-01-23:
I re-tested this on IOS 13.3 on an iPhone 6S, and it's still a problem. I can delete, say, `code.py` in the Files app, but when I browse away to another device and then browse back to CIRCUITPY, `code.py` is still there, and it's also still there if I remove the board and plug it into a regular host computer.

We could special-case the `.Trashes` folder and have anything moved into it automatically be deleted immediately. I'm not sure if that's a great user experience, but maybe it's no worse than the behavior of a `.Trashes` file.

ladyada, 2020-01-23:
like you mean circuitpy FAT handler would look for the path and delete any FAT entries?

dhalbert, 2020-01-23:
@ladyada, yeah, that's really kind of bad, and it would confuse the host computer, which would have a different idea of what the filesystem looks like. I take it back.

ladyada, 2020-01-23:
what if we have a trashes folder but its like, non-writable?

dhalbert, 2020-01-23:
> what if we have a trashes folder but its like, non-writable?

I'll try that.

dhalbert, 2020-01-23:
It turns out there are no per-entry file or directory permissions on ordinary FAT filesystems, so there's no way to make just that folder be read-only.

dhalbert, 2023-10-28:
I think we have this figured out now -- nothing to change.

closed 2023-10-28 by dhalbert