adafruit/circuitpython #2202: Add Native SSL Library

brentru, 2019-10-07:
We're currently using the mbedTLS implementation of the TLS and SSL protocols on the ESP32 Co-Processor/AirLifts by communicating with an ESP32 running [nina-fw](https://github.com/adafruit/nina-fw) from CircuitPython. While this solution is stable, a native CircuitPython SSL/TLS library would be faster, remove the hardware dependency for an ESP32 and allow for different transports (not just wifi).

Some embedded ssl/tls implementations:
BearSSL: https://bearssl.org
AxTLS: http://axtls.sourceforge.net

moto-timo, 2021-01-29:
Would be great to have a common TLS approach.

tannewt, 2021-02-01:
We could have the ssl module actually do SSL over the wrapped socket.

timonsku, 2022-01-26:
This would be great to have so Wiznet and potentially other Ethernet MACs like in the i.MX RT can support SSL too. Are there any plans yet?

askpatrickw, 2022-01-27:
There are MicroPython SSL implementations, could they be used as a starting point?

tannewt, 2022-01-27:
@timonsku No plans for Adafruit-funded folks.

@askpatrickw Yup! Definitely. Their code is usually well done and a good reference.

nabber00, 2023-06-12:
Seems like [WolfSSL Python](https://github.com/wolfSSL/wolfssl-py) should be pretty easy to add?

dhalbert, 2023-06-12:
> Seems like [WolfSSL Python](https://github.com/wolfSSL/wolfssl-py) should be pretty easy to add?

The WolfSSL library is GPL2 and commercial. We don't put GPL code in CircuitPython so as not to GPL-ize the whole thing, and the commercial restrictions are also an issue.

anecdata, 2024-05-17:
I think the CircuitPython core elements are in place for this, though it may need some changes in the ESP32SPI-related libraries. The native `ssl` module is only installed on [certain](https://github.com/adafruit/circuitpython/pull/8954#issuecomment-2062162495) boards though.

Tested config:

- UnexpectedMaker FeatherS3 `Adafruit CircuitPython 9.1.0-beta.2 on 2024-05-15; FeatherS3 with ESP32S3`
- Adafruit Airlift FeatherWing (NINA v1.7.1)

This code simulates an HTTPS Request over ESP32SPI using a native-`ssl`-wrapped socket:
```py
import time
import os
import board
import digitalio
import socketpool
import ssl
import adafruit_connection_manager
from adafruit_esp32spi import adafruit_esp32spi

HOST = "httpbin.org"
PATH = "/get"
PORT = 443

time.sleep(3)  # wait for serial

buf = bytearray(1024)

spi = board.SPI()
esp32_cs = digitalio.DigitalInOut(board.D13)
esp32_reset = digitalio.DigitalInOut(board.D12)
esp32_ready = digitalio.DigitalInOut(board.D11)
radio = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

pool = adafruit_connection_manager.get_radio_socketpool(radio)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(radio)

print(f'Connecting to wifi... ', end='')
radio.connect(os.getenv("WIFI_SSID"), os.getenv("WIFI_PASSWORD"))
print(f'{radio.ipv4_address}')

for _ in range(0, 10):
    print(f'{"-"*25}')
    print('Creating TCP client socket...')
    s = pool.socket(pool.AF_INET, pool.SOCK_STREAM)
    ss = ssl_context.wrap_socket(s, server_hostname=HOST)
    print('Connecting to remote socket...')
    ss.connect((HOST, PORT))

    sbytes = f'GET {PATH} HTTP/1.1\r\nHost: {HOST}:{PORT}\r\n\r\n'.encode()
    # note that esp32spi doesn't return the number of bytes sent like CPython:
    ssize = ss.send(sbytes)
    print(f'Sent {ssize} bytes: {sbytes}')

    rsize = ss.recv_into(buf)
    print(f'Received {rsize} bytes: {buf[:rsize]}')

    ss.close()
    time.sleep(1)
```
```
code.py output:
Connecting to wifi... 192.168.6.74
-------------------------
Creating TCP client socket...
Connecting to remote socket...
Sent None bytes: b'GET /get HTTP/1.1\r\nHost: httpbin.org:443\r\n\r\n'
Received 430 bytes: bytearray(b'HTTP/1.1 200 OK\r\nDate: Fri, 17 May 2024 15:47:20 GMT\r\nContent-Type: application/json\r\nContent-Length: 200\r\nConnection: keep-alive\r\nServer: gunicorn/19.9.0\r\nAccess-Control-Allow-Origin: *\r\nAccess-Control-Allow-Credentials: true\r\n\r\n{\n  "args": {}, \n  "headers": {\n    "Host": "httpbin.org", \n    "X-Amzn-Trace-Id": "Root=1-66477c07-2f1cad7c13d06e3f4d62294c"\n  }, \n  "origin": "108.74.220.152", \n  "url": "https://httpbin.org/get"\n}\n')
-------------------------
# etc...
```

anecdata, 2024-05-17:
I think the CircuitPython core elements are [now in place](https://github.com/adafruit/circuitpython/pull/8954) for this, but it needs some changes in the ESP32SPI-related libraries (and possibly NINA) to make the ESP32 sockets compatible. There's a compatibility checklist in #8954. The native `ssl` module is only installed on [certain](https://github.com/adafruit/circuitpython/pull/8954#issuecomment-2062162495) boards though.

Tested config:

- UnexpectedMaker FeatherS3 `Adafruit CircuitPython 9.1.0-beta.2 on 2024-05-15; FeatherS3 with ESP32S3`
- Adafruit Airlift FeatherWing (NINA v1.7.1)
- libraries are current release `...9.x-mpy-20240516`

This code tries to simulate an HTTPS Request over ESP32SPI using a native-`ssl`-wrapped socket (the socket comes from NINA, the wrapper comes from the ESP32-S3):
```py
import time
import os
import board
import digitalio
import socketpool
import ssl
import wifi
import adafruit_connection_manager
from adafruit_esp32spi import adafruit_esp32spi

HOST = "httpbin.org"
PATH = "/get"
PORT = 443

time.sleep(3)  # wait for serial

buf = bytearray(1024)

spi = board.SPI()
esp32_cs = digitalio.DigitalInOut(board.D13)
esp32_reset = digitalio.DigitalInOut(board.D12)
esp32_ready = digitalio.DigitalInOut(board.D11)
radio = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

radio2 = wifi.radio

pool = adafruit_connection_manager.get_radio_socketpool(radio)  # ESP32SPI
ssl_context = adafruit_connection_manager.get_radio_ssl_context(radio2)  # native wifi

print(f'Connecting to wifi... ', end='')
radio.connect(os.getenv("WIFI_SSID"), os.getenv("WIFI_PASSWORD"))
print(f'{radio.ipv4_address}')

for _ in range(0, 10):
    print(f'{"-"*25}')
    print('Creating TCP client socket...')
    s = pool.socket(pool.AF_INET, pool.SOCK_STREAM)
    ss = ssl_context.wrap_socket(s, server_hostname=HOST)
    print('Connecting to remote socket...')
    ss.connect((HOST, PORT))

    sbytes = f'GET {PATH} HTTP/1.1\r\nHost: {HOST}:{PORT}\r\n\r\n'.encode()
    # note that esp32spi doesn't return the number of bytes sent like CPython:
    ssize = ss.send(sbytes)
    print(f'Sent {ssize} bytes: {sbytes}')

    rsize = ss.recv_into(buf)
    print(f'Received {rsize} bytes: {buf[:rsize]}')

    ss.close()
    time.sleep(1)
```
```
code.py output:
Connecting to wifi... 192.168.6.74
-------------------------
Creating TCP client socket...
Traceback (most recent call last):
  File "code.py", line 38, in <module>
AttributeError: 'Socket' object has no attribute 'type'
```

jepler, 2024-06-14:
It's now possible to use the core ssl module with sockets implemented in Python (#8954); this actually works with wiznet. However, nobody's checked whether this actually works with airlift; it might require changes in the airlift socket implementation to correctly align with the standard socket. If it doesn't, the esp32spi github repo is probably the right place to open issues.

Additionally, few or no builds for board with an airlift include the core ssl module. If there are combos for which this is sensible, a PR within CircuitPython to enable SSL would be the way to go.

closed 2024-06-14 by jepler