adafruit/circuitpython #2203: wav files on CPXb crash on fifth button touch

gallaugher, 2019-10-07:
I've tried modifying the code for playing wav files, but notice on the fifth button press things seem to crash, with a buzzing sound from the speaker & the sound playing continuously. Sprinkling in print statements shows it hits audio.play(wave) before choking, so maybe this a bug in the current libs? Ctrl-C stops in the while audio.playing: loop. Here's a GitHub repo that has the short file (detects a button A press as well as A1 cap touch), and plays two sounds based on the current CPXb tutorial on Adafruit: 
https://github.com/gallaugher/CPXBluefruitHalloweenHead

And if it's useful, here's a subset of the code w/function + while True: loop
def play_file(filename):
    wave_file = open(filename, "rb")
    with WaveFile(wave_file) as wave:
        with AudioOut(board.SPEAKER) as audio:
            audio.play(wave)
            while audio.playing:
                 pass
    print("Finished")

while True:
    if touch_A1.value:
        pixels.fill(ORANGE)
        play_file("evil_laugh.wav")
        pixels.fill(BLACK)
        time.sleep(0.5)

    if button_A.value:
        pixels.fill(RED)
        play_file("laugh.wav")
        pixels.fill(BLACK)
        time.sleep(0.5)

ladyada, 2019-10-07:
@jepler this is a good one for ya when you return!

jepler, 2019-10-07:
I should be able to look at this someone next week

jepler, 2019-10-12:
Reproduced on my own CPB.  My reproducer script is:
```!python
def count():
    i = 0
    while True:
        yield i
        i += 1

from audiopwmio import PWMAudioOut as AudioOut
from audiocore import WaveFile
wf = open('cymbaly-1.wav', 'rb')
wave = WaveFile(wf)
import board

for i in count():
    print(i)
    with AudioOut(board.SPEAKER) as audio:
        print(audio)
        audio.play(wave)
        while audio.playing: pass
    print("Finished")
        
``` 
`cymbaly-1.wav` is a random wave file with these characteristics: `RIFF (little-endian) data, WAVE audio, Microsoft PCM, 16 bit, mono 16000 Hz`

ladyada, 2019-10-12:
[circuitplayground_bluefruit.zip](https://github.com/adafruit/circuitpython/files/3721025/circuitplayground_bluefruit.zip)

here's the artifacts if ya dont want to wait for a release!

closed 2019-10-12 by ladyada

gallaugher, 2019-10-19:
Thanks @ladyada , I'd love to use a working version w/students on Monday, but when I download the .zip above, then grab the .uf2 in en_US (adafruit-circuitpython-circuitplayground_bluefruit-en_US-20191012-a146ce5.uf2), then delete the old .uf2 from the CPXb and copy this new file over, unplug & plug back in, then run @jepler's sample code (using my own .wav), I still get the same behavior - a buzzing hang instead of a 5th play. So sorry if I'm doing something obviously wrong. Thanks!

ladyada, 2019-10-19:
please try the latest build on S3 in case it makes a difference

gallaugher, 2019-10-19:
So sorry - @jepler set me straight. I was copying over the .uf2 you'd kindly linked above, but to CIRCUITPY, I didn't double-click to copy it to CPLAYBOOT. Once I did that all worked great. Didn't mean to trouble you. This'll be great to introduce on Monday. I very much appreciate your kind help. Cheers!

ladyada, 2019-10-19:
awesome!

PR #2212 (2019-10-12): nrf: PWMAudioOut: deactivate PWM when deinitting self
Otherwise, when an AudioPWMOut object was deinitted without being explicitly `stop()`ped, it would use up a slot in active_audio[]; the 5th iteration would create a non-working audio object which would just buzz instead of playing the right thing.

(In a separate commit, stylistic changes are made to better match the CP coding style)

Closes: #2203
commit fae6e295 (2019-10-12): nrf: PWMAudioOut: deactivate PWM when deinitting self

.. otherwise, when an AudioPWMOut object was deinitted without being
explicitly stop()ped, it would use up a slot in active_audio[]; the
5th iteration would create a non-working audio object which would just
buzz instead of playing the right thing.

Closes: #2203