adafruit/circuitpython #2221: bleio

tcpipchip, 2019-10-15:
[image]
and i am doing something wrong ?

dhalbert, 2019-10-15:
If you are using the latest alpha on an nRF52840 board, then you need the latest prerelease of the `adafruit_ble` library: https://github.com/adafruit/Adafruit_CircuitPython_BLE/releases/tag/2.0.1-alpha.2

tcpipchip, 2019-10-15:
yes, i was! I wil test

tcpipchip, 2019-10-15:
how can i see the advertsing data ?
[image]

tcpipchip, 2019-10-15:
found...
print(e.manufacturer_specific_data)
But it´s returning always
**None**

tcpipchip, 2019-10-15:
i tested doing a ARDUINO BLE SCANNER

dhalbert, 2019-10-15:
Moving this as an issue on the library. Will investigate.