adafruit/circuitpython #2225: SPI slave and I2C slave for SAMD51 boards

jstanle1, 2019-10-18:
Please implement SPI slave and I2C slave for the SAMD51 and SAMD21 boards.  The SPI slave would benefit from being implemented using DMA, as well.

Thanks.

dhalbert, 2019-10-18:
It's already implemented, but not turned on on the sAMD21 boards due to lack of space. You could turn it on and turn something else off. It's controlled by the `CIRCUITPY_I2CSLAVE` value, which you can set in  `mpconfigboard.mk`. E.g., look at `ports/atmel-samd/boards/feather_m0_express/mpconfigboard.mk`

dhalbert, 2019-10-18:
https://circuitpython.readthedocs.io/en/4.x/shared-bindings/i2cslave/I2CSlave.html

jstanle1, 2019-10-18:
Dan -- thanks very much for you replies.  Any word on whether this might work for the SAMD51 boards?

dhalbert, 2019-10-18:
This was indeed tested on SAMD51 boards when the PR was done: https://github.com/adafruit/circuitpython/pull/1064#issuecomment-410435059. It is included in the SAMD51 builds.

jstanle1, 2019-10-18:
Great, thanks again for the word on ths.

closed 2019-10-18 by dhalbert