adafruit/circuitpython #2231: particle:argon uf2 installation

WilliamGFish, 2019-10-18:
With my Argon the CircuitPython installation has never worked.

I have reinstalled the boot-loader several times but the uf2 file after writing fails to have any effect.

Help please..

Billy..

WilliamGFish, 2019-10-18:
I have followed the instructions and my boron lte board is the same

tannewt, 2019-10-18:
Are you able to see the *BOOT drive after installing the bootloader?

WilliamGFish, 2019-10-18:
Yeah the bootloader shows as ARGONBOOT and BORONBOOT as expected. When I drop the uf2 file onto drive it writes then nothing. No drive appears but the COM port is active.

tannewt, 2019-10-18:
Try listing the drive from the REPL. `os.listdir()` Did you just get the Particle boards? It's possible they switched SPI flash chip and CircuitPython isn't configured to talk with it.

WilliamGFish, 2019-10-18:
OK now i'm not sure what to do... REPL- I've got Mu which just tells me there is not Adafruit boards available.
The Particle boards are built around an NRF52840 whicih I have been using with Zephyr for months but am currently trying to re-program (re-flash to stock) the ESP32 for standard AT commands.

Circuit Python was a good choose to do this but I'm failing at the first hurdle.

tannewt, 2019-10-18:
You can access the REPL without Mu by following these instructions: https://learn.adafruit.com/welcome-to-circuitpython/kattni-connecting-to-the-serial-console#using-something-else-8-11

You'll have to look into the Particle docs to find out if there is a different SPI Flash.

WilliamGFish, 2019-10-18:
They both use the MX25L1606. I have attached block diagrams

[image]
[image]

WilliamGFish, 2019-10-18:
I use Putty and Termite but just get a blank screen. This is okay, according to the guide? 

How do I get my program to the board? Do I go back to BOOT mode them copy?

WilliamGFish, 2019-10-18:
PROGRESS:
Just flashed with the Adafruit Feather nRF52840 Express with nRF52840 version of CircuitP and we have REPL prompt

WilliamGFish, 2019-10-18:
PROGRESS:
Used _**adafruit-circuitpython-pca10059-en_US-4.1.0.uf2**_ and it has appeared as drive and COM port

WilliamGFish, 2019-10-19:
I have been following these instructions and not having success

jerryneedell, 2019-10-19:
Just to clarify - did you try the adafruit_circuitpython_particle_argon_en_US-4.1.0.uf2?

WilliamGFish, 2019-10-19:
when I use ARGON version doesn't appear as a USB drive.

If I use Xenon then USB Drive (CURCUITPY) appears

WilliamGFish, 2019-10-19:
With ARGON or BORON still nothing with XENON:

```
Adafruit CircuitPython 4.1.0 on 2019-08-03; Particle Xenon with nRF52840
>>>
>>>
>>>
>>>
>>>
>>> control=C
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'C' is not defined
>>>

```

The only thing that separates them is the UART connection to Modems

Any thoughts?

jerryneedell, 2019-10-19:
Makes no sense to me.  I use the argon build on argon and xenon build on xenon. 
Not sir3 if it matters, do you have the argon boot loader on the Argon? How did you install the boot loader?  Did you use this guide?   https://learn.adafruit.com/circuitpython-on-the-nrf52/build-flash-particle

WilliamGFish, 2019-10-19:
That is the guide is used and have used the "correct" version on both the Argon and Boron boards with the same result.

I have been using these boards with Zephyr for months now and was looking at circuitpython as a alternative but its not been successful so far.

WilliamGFish, 2019-10-22:
I'm back on this and still not able to either my Argon or Boron board to bring up Circuit Python?

I have tried various boot-loaders and uf2 but still nothing.

If a flash the uf2 for the Xenon on either a Argon or Boron it comes up straight away.

Lost....need some ideas.

Billy..

WilliamGFish, 2019-10-22:
Some direct reponces to questions:

**> Do you have the argon boot loader on the Argon?** 
I'm using the Argon on the Argon and Boron on the Boron.

adafruit-circuitpython-particle_argon-en_US-4.1.0.uf2
adafruit-circuitpython-particle_argon-en_US-4.0.1.uf2
adafruit-circuitpython-particle_argon-en_US-4.0.0.uf2
adafruit-circuitpython-particle_argon-en_US-5.0.0-alpha.4.uf2

adafruit-circuitpython-particle_boron-en_US-4.1.0.uf2

**> How did you install the boot loader?** 

**> Did you use this guide?** https://learn.adafruit.com/circuitpython-on-the-nrf52/build-flash-particle
Yes that was my starting point, yet i'm using pre-compiled versions and boot-loader and uf2

jerryneedell, 2019-10-22:
How did you flash the bootloader -- with the Arduino IDE?
get your board into the Bootloader and post the contents of the INFO_UF2.TXT file.

jerryneedell, 2019-10-22:
I just flashed the afadfruit_bluefruit_nrf52840 bootlaoade via a J-Link as described in the guide and then put the particle_argon_4.1.0.uf2 on it and all worked fine.

WilliamGFish, 2019-10-22:
> How did you flash the bootloader
I used nrfjprog, I even out of habit do an eraseall.

`nrfjprog --program particle_argon_bootloader-0.2.13_s140_6.1.1.hex --sectoranduicrerase -f nrf52 --reset`
> get your board into the Bootloader and post the contents of the INFO_UF2.TXT file.
The bootloader works as expected which is the frustrating thing.

I am starting to think its a windows thing, Win10 can be a pain "resolving" problems for you.

Thanks for your continued support.

Billy..

jerryneedell, 2019-10-22:
If you have arduino installed -- update to the lated adafruit nrf5284 BSP and try that.

WilliamGFish, 2019-10-22:
> If you have arduino installed -- update to the lated adafruit nrf52840 BSP and try that.

Not an Arduino guy, normally i would be hacking code onto the NRF52 via C and nrfjprog. I generally don't use them USB drives, USB is just as a serial console.

Thats why I'm startting to suspect this could be a windows problem

jerryneedell, 2019-10-22:
ddi you do all three steps 
[image]

jerryneedell, 2019-10-22:
I may be wrong -- where did you get the bootloader file -- I will try to replicate it.

jerryneedell, 2019-10-22:
OK -- I just downloaded the same bootloader .hex file from the nrf52 bootloader repository and used the same command to flash it (after erasing) and then loaded the 4.10. .UF2 and it all worked as expected. perhaps it it is windows -- I'm out of ideas ....

WilliamGFish, 2019-10-22:
Jerry, found it.... ruddy Windows. 
In device manager it was only showed as Argon serial port. Not normally a problem as I only use serial comms. 
So un-installed device and started again...bingo it's working. 

Now how do we get this into the trouble shooting guide.

jerryneedell, 2019-10-22:
Congratulations! Glad it is working -- I think @dhalbert maintains that guide.

dhalbert, 2019-10-22:
Is this Windows 10 or 7? Did you install any drivers initially that might have taken control of this? I don't remember if Particle supplies any drivers.

jerryneedell, 2019-10-22:
Still odd that you had partial success with the xenon build ....

WilliamGFish, 2019-10-22:
The Xenon build presented a difference device which appeared as a USB Drive, which is the clue I needed to trace the issue. 100% Windows trying to be clever but failing.

I don't use a Particle Xenon as I use the NRF52_DevKit seems silly to buy another version of the same thing.

So if someone does not see a CIRCUITPY drive they need to look in Device Manager -> Disk Drives; if they do not see an Argon/Boron/Xenon drive they need to check ->Ports (COM &LPT) to see if they have ARGON Serial (or corresponding version). Then right click and uninstall device (including drivers). 
Plug board back in and it should configure correctly.

closed 2019-10-31 by WilliamGFish

jrbski, 2020-05-24:
> The Xenon build presented a difference device which appeared as a USB Drive, which is the clue I needed to trace the issue. 100% Windows trying to be clever but failing.
>
> I don't use a Particle [...]

I had the same issue, and the part I was missing was the 'including drivers' part of uninstalling the serial port driver (in my case for a xenon) under win10.. once I hit the checkbox to remove the drivers, the circuitpy drive appeared.

Thanks @WilliamGFish for the great summary on this issue!