adafruit/circuitpython #2233: Add TC74 i2c temperature sensor library

iayanpahwa, 2019-10-20:
TC74 from microchip is a cheap and good i2c based temperature sensor, will be good to have a supported library in CircuitPython

ladyada, 2019-10-20:
this will probably work https://github.com/adafruit/Adafruit_CircuitPython_PCT2075

iayanpahwa, 2019-10-21:
Thanks @ladyada , I'll give it a try .

ladyada, 2019-10-21:
@siddacious @kattni plz update the library readme to clarify it will also work on the TC74 :)

iayanpahwa, 2019-10-21:
@ladyada do you think we can update the library to be generic and used with both. Not sure but TC74 supports low power modes, can add function to utilise that, I can work on it. :)

ladyada, 2019-10-21:
thats right, some things may vary, please check the TC74 and PCT2075 datasheet, i believe PCT has everything TC74 does but please add any functions!

iayanpahwa, 2019-10-21:
sure

siddacious, 2019-10-22:
@ladyada README updated!