adafruit/circuitpython #2253: nrf I2C hang

dhalbert, 2019-10-29:
If an nRF I2C (TWIM) peripheral is connected to pullups that are not pulled up, and then connected to an unpowered device, the peripheral can be initialized, but then operations on the bus may hang.

First noted on Arduino Nano 33 BLE. It has an on-board accelerometer. The I2C pullups and the device are both powered from other pins, so they can be shut off when not in use. If the pullups and the device are unpowered, we can still create a `busio.I2C()`, because it passes the tests that try to detect whether pullups are connected, but then trying to read from the device will hang CircuitPython.

This can be simulated on a Feather nRF52840 by putting a 10K resistor between SCL and SDA and then connecting an unpowered device (I used an LIS3DH) to SCL and SDA. Trying to `readfrom_into()` hangs the code in a busy-wait loop:

https://github.com/adafruit/nrfx/blob/8935519a689c88962b84c661ac606e555898b81a/drivers/src/nrfx_twim.c#L459
```
        while (!nrf_twim_event_check(p_twim, evt_to_wait))
```

Some possible solutions are:

1. Improve how to check for proper pullups: maybe try using internal pullups and pulldowns. This doesn't solve the problem of losing pullup power later, for instance.
2. Add a timeout of some kind to `nrf_twim_event_check()`, directly, or as a background task.

What might be a good timeout (several seconds??), and should it be specifiable in the I2C constructor?

We are using the synchronous TWIM driver, but going to the async driver would not help, I think, because the IRQ handler is not going to get called. So we still need to implement a timeout mechanism. The current Arduino BSP for nRF52 uses the async driver, but it still busy waits and I think it may have the same problem.

Need to decide how much effort to expend to fix this vs. just documenting it.

Tagging @ladyada who discovered this issue.

dhalbert, 2019-10-29:
The primary problem for @ladyada was that if `code.py` got into this state quickly, it seemed impossible to get into safe mode. Perhaps we just need an easier way to get into safe mode. Or perhaps pressing reset during certain operations might initiate automatic safe mode (e.g., set a flag before I2C operations and unset it when done; the flags indicates one should reset into safe mode and not restart).

ladyada, 2019-10-29:
yes improved safemode would be great... 
but also lets have a timeout of 1 second, that's forever in i2c bus time :)

tannewt, 2019-11-01:
@ladyada Did you try the slow double tap to get into safe mode? CircuitPython should wait 700ms on start up to see if a reset occurs and then enter safe mode if it does. Its the same trick as the bootloader, just a wait after the bootloader's wait.

ladyada, 2019-11-01:
i tried but was not successful - i probably didnt have the timing right

dhalbert, 2020-02-20:
Related to #2635. Deferring until after 5.0.0.

dhalbert, 2023-06-23:
@furbrain Moving discussion from #8093 here:

My idea above was to modify `nrfx/drivers/src/nrfx_twim.c` to include a timeout in the synchronous transaction busy-wait loops like you see in the original post. I never got around to this, but it's not such a big deal: we already have our own patches in our fork of that library. I could put this on the list for 9.0.0. I would also check whether Nordic has already done something about this, because I think there were other people with the same issue.

furbrain, 2023-06-23:
Ah ok, I was going to try and avoid modifying the nrfx library as it was
third party.

furbrain, 2024-01-23:
Do we have any thoughts about how long the timeout should be? Maybe calculate how long the transfer should take, then double it? and then add say 5ms  for shorter transfers?

dhalbert, 2024-01-23:
The main point of the timeout is to catch a bus that's hanging. I don't think we need to calculate anything. So 1 second is probably more than enough, as @ladyada mentioned above.

ladyada, 2024-01-23:
1 second plz

furbrain, 2024-01-23:
Looking at the code, maximum data that can be transferred  in "one go" is 65k - at 100kHz and assuming 10 bits per byte (covers ACK etc), this would take 6.5s. So a 1s timeout may incorrectly fail.

Choices would be to have a computed timeout, or reduce the chunk size to say 1k - I'd be inclined to go with the latter

PR #8827 (2024-01-25): Fix I2C hang when power removed from pull-ups
This PR fixes #2253 and fixes #8093.

### Explanation
This works by creating the nrfx_twim device with a callback (``twim_event_handler``). 
When the callback is triggered it marks that the transfer has finished and records any error/success code in the twim_peripheral data structure.
We replace the nrfx_twim_xfer function with our ``_twim_xfer_with_timeout`` which make a note of the start time, and waits until either the callback has triggered or 1s has elapsed.
If the timeout is triggered it generates an ``OSError: Timed out``

Trying to use the  [... cut]
closed 2024-01-25 by dhalbert