adafruit/circuitpython #2255: Reported bugs in nrf i2s audio implementation

jepler, 2019-10-30:
Github user @rdagger reports the following problems:

> I've been having problems with I2S on the Feather nRF52840 Express running CircuitPython 5.0.0-alpha.4-260-g2b02750f1 (2019-10-26.)
> I'm using an Adafruit I2S 3W Class D Amplifier Breakout - MAX98357A (PRODUCT ID: 3006).
> I've been using the example programs and here are some issues:
> 1.  The sine waves only work well at certain frequencies.  Some frequencies causes the Feather to reboot in safe mode.  Other frequencies have very rough waves.
> 2.  Often after an I2S program completes without errors, the Feather will reboot a few minutes later.
> 3.  WAV files have to have a frequency of 22050Hz and 16 bits per sample or else they are garbled.

Look into these reported issues and fix them as necessary.

Note: I am self-assigning this so it stays on my radar, but other obligations will likely prevent me from doing more than the smallest amount of investigative work until at least the week of November 11.

jepler, 2019-11-25:
I reproduced a crash using a frequency of 727.  ```#0  reset_into_safe_mode (reason=reason@entry=HARD_CRASH)
    at ../../supervisor/shared/safe_mode.c:84
#1  0x0004373c in HardFault_Handler () at supervisor/port.c:168
#2  <signal handler called>
#3  i2s_buffer_fill (self=0x20008700) at common-hal/audiobusio/I2SOut.c:143
#4  0x00008000 in ?? ()
```
Somehow, rather than copying 22 bytes and stopping, the copy runs past the end of memory and hard crashes:
```139             uint16_t *bp = (uint16_t*)buffer;
140             uint16_t *be = (uint16_t*)(buffer + bytecount);
141             uint16_t *sp = (uint16_t*)self->sample_data;
142             for (; bp != be; bp++) {
143                 *bp++ = *sp++ + 0x8000;
144             }
```
.. did you spot the error?  whoops.

jepler, 2019-11-25:
@rdagger can you provide an example wav file which played incorrectly for you?

jepler, 2019-11-25:
MAX98357A requires samples be 16, 24, or 32 bits.  A sample of 8 bits won't work properly; due to a bug they are currently played at double the frequency and sound is produced by the breakout; after fixing the bug the waveform is correct, but no sound is produced.

Besides that, even common 22050Hz and 44100Hz samples are technically out of spec, but do seem to work (only 8, 16, 48, and 96kHz  +- 5% sample rates are permitted per the datasheet)

[image]

jepler, 2019-11-25:
On the other hand, the UDA1334 I2S DAC *can* play 8-bit samples properly

rdagger, 2019-11-25:
Here's a sample WAV that produced nothing but noise.
[mi-destruct.zip](https://github.com/adafruit/circuitpython/files/3888887/mi-destruct.zip)

I tried again today with the latest build and it was intelligible but the speed was too fast.

rdagger, 2019-11-25:
I haven't done much testing but here is what works and does not work for me:

**Not Working:**
Sample number:		29334
Total duration:		00:00:02:660
Frequency:		11025Hz
Bits per sample:	8
Format:			PCM

**Working:**
Sample number:		168960
Total duration:		00:00:07:662
Frequency:		22050Hz
Bits per sample:	16
Format:			PCM

jepler, 2019-11-29:
@rdagger Unfortunately, it's a hardware limitation that MAX98357A doesn't support 8-bit samples like the one in mi-destruct.zip.  We might address this in software in the future (#2323), but for now it is not going to work.

A tool like audacity can convert wave files, though obviously the flash storage usage is higher for 16 bits.

I'm going to close this up for now, but please feel free to open fresh issues if there are problems I haven't covered yet.  Thanks for working with us on this!

closed 2019-11-29 by jepler

rdagger, 2019-11-29:
An error message defining the hardware limitations would be helpful.  Thanks.

ladyada, 2019-11-29:
we can't know theres an error because the hardware limitation is with the i2s chip NOT the circuitpython core! other i2s chips may work :)