adafruit/circuitpython #2258: Disable CircuitPlayground Bluefruit speaker by default

tannewt, 2019-11-01:
We need to disable the speaker on the CircuitPlayground Bluefruit just like we do on the Express because it currently makes noise when the AUDIO pin is touched.

jepler, 2019-11-01:
A little quick analysis: It looks like the related code in the samd port for the express version is:
```
    #ifdef SPEAKER_ENABLE_PIN
    speaker_enable_in_use = false;
    gpio_set_pin_function(SPEAKER_ENABLE_PIN->number, GPIO_PIN_FUNCTION_OFF);
    gpio_set_pin_direction(SPEAKER_ENABLE_PIN->number, GPIO_DIRECTION_OUT);
    gpio_set_pin_level(SPEAKER_ENABLE_PIN->number, false);
    #endif
```
(reset_all_pins)
and a few other bits all gated by SPEAKER_ENABLE_PIN.  Similar code will need to be adapted into the nRF port, and SPEAKER_ENABLE_PIN provided/defined.

kevinjwalters, 2019-11-01:
How are you observing the noise, on VBUS and VCC or is this also making RFI? Specifically, I'm assuming this is actually the joy of class D amps and is coming from the PAM8301?

tannewt, 2019-11-01:
Yup, just hear it out of the speaker when I touch the audio pad. It's not the end of the world but it is unexpected.

kevinjwalters, 2019-11-01:
I was interested in anything PAM8301 related because I somehow let the magic smoke out of mine with some rather innocent use, I added some notes to an existing thread: [Adafruit Forums: Circuit Playground Express gets VERY hot](https://forums.adafruit.com/viewtopic.php?f=58&t=157190).

PR #2262 (2019-11-03): cpb: Disable the onboard speaker until request
This allows the board to disable the onboard speaker until explicitly enabled in user code.
    
Testing performed on a CPB:
 * Touching the AUDIO pin with a fingertip no longer generates noise/buzz
 * Generating a waveform with `simpleio.tone` produces no sound by default
 * When the board.SPEAKER_ENABLE is configured as a digital output and
       set True, `simpleio.tone` does produce sound
    
Note that while guides should include information about SPEAKER_ENABLE, it's possible that some users who omitted it could view this as a breaking chan [... cut]
commit 2bdccf03 (2019-11-03): cpb: define the SPEAKER_ENABLE_PIN

This allows the board to disable the onboard speaker until explicitly
enabled in user code.

Testing performed on a CPB:
 * Touching the AUDIO pin with a fingertip no longer generates noise/buzz
 * Generating a waveform with `simpleio.tone` produces no sound by default
 * When the board.SPEAKER_ENABLE is configured as a digital output and
   set True, `simpleio.tone` does produce sound

Note that while guides should include information about SPEAKER_ENABLE, it's
possible that some users who omitted it could view this as a breaking change.
They can fix it by  [... cut]
closed 2019-11-03 by dhalbert