adafruit/circuitpython #2267: os.stat() causing int overflow on boards without long int support

caternuson, 2019-11-04:
Ran into this here:
https://github.com/adafruit/Adafruit_CircuitPython_framebuf/issues/25

Can recreate by trying this on any file:
```python
Adafruit CircuitPython 4.1.0 on 2019-08-02; Adafruit Trinket M0 with samd21e18
>>> import os
>>> os.stat("/foo.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: small int overflow
>>> 
```

dhalbert, 2019-11-05:
This overflow is due to #1143, which restored using the 1970 epoch to `os.stat()`, so that the timestamps returned would match standard Unix timestamps. Previously it was using a year 2000 epoch.

We could use year 2000 epoch timestamps for non-longint builds, or we could just return 0, say. @notro do you have an opinion or suggestion?

Note that on non-longint builds, `time.time()` and other epoch-1970 related functions are turned off (see #846, which switched back to 1970 epoch for other `time` functions).

dhalbert, 2020-02-06:
My inclination is to "fix" this just by returning 0, which will be 1969-12-31 midnight. The time functions I mentioned in #846 throw exceptions when the time is out of range, but that seems like a bad idea for `os.stat()`. We can document the 1969 behavior. If no objections, I'll do this for 5.0.0.

tannewt, 2020-02-06:
0 is fine with me!

PR #2626 (2020-02-18): avoid os.stat() int overflow on smallint-only builds
Fixes #2267.

Returns the number of seconds for 1999-12-31 for `os.stat()` fields that are times, on smallint builds only. This is instead of 1969-12-31. 1999-12-31 is used for some other fixed dates, so using it is a bit more consistent.
closed 2020-02-18 by tannewt