adafruit/circuitpython #2278: presence build fails on RPi

jerryneedell, 2019-11-09:
I've been testing building CP on various platforms -- the current master builds fine for all BSPs on Ubuntu Linux and on MacOS. On my Raspberry Pi 4  all of the boards build OK except for the spresense - it fails with:

```
pi@gjnpi4desk:~/projects/circuitpython/ports/cxd56 $ make BOARD=spresense
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
GEN build-spresense/genhdr/moduledefs.h
QSTR updated
AR build-spresense/libmpy.a
LD build-spresense/firmware.elf
Creating build-spresense/firmware.spk
spresense-exported-sdk/sdk/tools/linux/mkspk -c 2 build-spresense/firmware.elf nuttx build-spresense/firmware.spk
spresense-exported-sdk/sdk/tools/linux/mkspk: 1: spresense-exported-sdk/sdk/tools/linux/mkspk: Syntax error: "(" unexpected
make: *** [Makefile:205: build-spresense/firmware.spk] Error 2
```
I've just started looking into to it , but wanted to post this to see if anyone else has seen this or has any suggestions ...

jerryneedell, 2019-11-09:
ah -- this may be the problem:
```
pi@gjnpi4desk:~/projects/circuitpython/ports/cxd56/spresense-exported-sdk/sdk/tools/linux $ ./mkspk 
-bash: ./mkspk: cannot execute binary file: Exec format error
```

jepler, 2019-11-09:
Based on the information available, this is an issue to pursue upstream with https://github.com/sonydevworld/spresense-exported-sdk -- a binary executable program included in spresense-exported-sdk is only for "x86_64" linux systems, so it won't work on ARM based system such as rpi:

```
$ file ./sdk/tools/linux/mkspk
./sdk/tools/linux/mkspk: ELF 64-bit LSB executable, x86-64, version 1 (SYSV),
 dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32,
 BuildID[sha1]=0df49f2269405eaa8872ff7244d485c45f702703, not stripped
```

tannewt, 2019-11-11:
@kamtom480 Any suggestion?

kamtom480, 2019-11-11:
Unfortunately, currently mkspk on ARM is not supported. In the next Spresense SDK release, it will be possible to support this. When it is released, I will prepare PR.

jerryneedell, 2020-01-14:
tested after #2507 -- builds OK on Raspberry Pi 4
Thank you!

closed 2020-01-14 by jerryneedell