adafruit/circuitpython #2279: IS25LP128F Flash Problem

Casey10110, 2019-11-10:
Hi Guys,

I've got a board based on the Feather M4 Express that uses a 16 MBit IS25LP128F flash chip. Although it seems to be recognized in the JEDEC ID check, it hangs right in the spot shown in the following code and I am not quite sure why. It would be good to figure this out so that others don't assume that this chip will work in their design (I actually also couldn't get the W25Q128JV_SQ to work either, and that was even harder, where it looked like the wrong size was reported etc (see [https://forums.adafruit.com/viewtopic.php?f=60&t=154512](url)). Here is the piece of code that it hangs in in Adafruit_SPIFlash.cpp:

`  // The write in progress bit should be low.
  while ( readStatus() & 0x01 ) {}
	Serial.println("YOU SEE THIS ...");
  // The suspended write/erase bit should be low.
  while ( readStatus2() & 0x80 ) {}
	Serial.println("BUT DON'T SEE THIS ...");
  _trans->runCommand(SFLASH_CMD_ENABLE_RESET);
  _trans->runCommand(SFLASH_CMD_RESET);`

deshipu, 2019-11-11:
Do you have a schematic of how the flash chip is connected?
Do you happen to have a logic analyzer and can look at the signals?

Casey10110, 2019-11-11:
I can't post the entire schematic, but it is connected exaclty the same as on the Feather M4 in both cases (W25Q128JV_SQ and IS25LP128F). 

[image]

I don't have a logic analzer, but the fact that the flash_info.ino recognizes a chip shows that they are extremely likley connected correctly. In fact, another board with the exact same design and a different chip works perfectly (same chip as on Feather M4).

deshipu, 2019-11-11:
Did you try it in a regular SPI mode, not QSPI maybe?

tannewt, 2019-11-11:
@Casey10110 are you using CircuitPython or Arduino to test this?

Casey10110, 2019-11-20:
> @Casey10110 are you using CircuitPython or Arduino to test this?

I've actually tested on both and neither works. I'd warn people against using any 16 MB chips until this is solved!

Casey10110, 2019-11-21:
I'd recommend taking all three of these chips of the devices.h list at this point unless someone actually got them to work.

arturo182, 2019-11-21:
Hi, I'm the one who added the IS25LP128F definition, which I verified with my HW. Please don't blame the SW for your HW problems :)

https://github.com/adafruit/circuitpython/pull/2219

Casey10110, 2019-11-21:
For sure, Arturo, I don't want to blame the software. The thing is, the hardware setup is EXACTLY the same as for the other chips that work on the EXACT same board. If you want me to send you a schematic (or even two boards, one with a chip that works, and another that doesn't) to verify this I will. Others will have this issue for sure.

Casey10110, 2019-11-26:
An update: I got everything to work using a 8 MBit flash chip instead on the same board. So there is something wrong with the software for all the 16 MBit chips ... at least with my setup. It doesn't make sense that it would be a hardware issue when I can get multiple other chips to work on the exact same board, literally just swapping out. Thank you everyone for your help with this. I hope we can eventually get the other larger chips working for this board (16MBit is actually the maximum you can use for SAMD51 according to datasheet because of way memory is mapped).

jepler, 2021-07-24:
Closing because we have reports of IS25LP128F working on Arduino Nano RP2040 boards and current versions of CircuitPython.

closed 2021-07-24 by jepler

jepler, 2021-07-25:
(#5055)