adafruit/circuitpython #2284: allow stm32f405  to run at 168mhz

manitou48, 2019-11-13:
in  ports/stm32f4/peripherals/stm32f4/stm32f405xx/clocks.c
increase CPU clock from 84mhz to 168mhz with this change
**RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;**

dhalbert, 2019-11-13:
@hierophect I guess this would affect the mp_hal_delay_us calculation too.

hierophect, 2019-11-13:
I would consider this a part of a planned clock settings rework, which should ideally calculate this and other values from settings in `mpconfigboard`. It also needs to take into account that oscillators for the ST F405 boards vary from 8MHz to 12MHz. I can up the processor speed in the short term without doing the full rework, but I'll want to make sure it doesn't functionality of SPI and other peripherals, and it'll raise their minimum speeds, which may prohibit use with certain off chip modules.

manitou48, 2019-11-13:
Sounds like a plan.  Note that if I upload micropython on to Adafruit STM32F405, it runs at 168mhz, and if I use the Arduino IDE, sketches run @168mhz on the STM32F405.  micropython can change the CPU and bus speeds from python with pyb.freq()

I may try changing the divisor in circuitpython and see what breaks.  I haven't tried building a circuitpython .bin, but back in 2015 I was building micropython for pyboard.

hierophect, 2019-11-13:
@dhalbert do we support CPU and bus speed adjustment from python? I imagine it's less relevant for the NRF and Atmel ports since they assume set frequencies, but it'd be handy for the STM32.

ladyada, 2019-11-13:
as an aside, i would like to have samd51 overclocking support - it overclocks nicely and you get quite a boost! so if y'all come up with a clock adjustment protocol, id want to have it available for other chips

manitou48, 2019-11-13:
i changed clocks.c to use DIV2 and ran some circuitpython tests with time.sleep(), time.monotonic() and with a scope.  stm32f405 seems to be running at 168MHz.  I didn't find any problems.  But absence of proof is not proof of absence. :-)

hierophect, 2019-11-13:
I've also tested it with a variety of modules today. Submitting a PR momentarily

hierophect, 2019-11-13:
It's the same bus frequencies anyway due to the APB divisors - I thought there was a APB divisor quirk that would get in the way but it's only at certain low divisor values, and doesn't apply to SPI, etc. So minimum speeds will be unchanged.

hierophect, 2019-11-13:
@ladyada I've found some interesting articles about STM32 overclocking today too so that might be a future PR for fun and profit.

tannewt, 2019-11-14:
For general clock changing we could just make `microcontroller.cpu.frequency` settable. We'd need to validate that all of the downstream clocks are ok when this is done though.

manitou48, 2019-11-14:
>  We'd need to validate that all of the downstream clocks are ok when this is done though.

For the pyboard (STM32F405) and micropython's pyb.freq(), changing CPU and bus frequency is managed in stmhal/modmachine.c, machine_freq()

closed 2019-11-15 by dhalbert

dewalex, 2021-03-04:
> For general clock changing we could just make `microcontroller.cpu.frequency` settable. We'd need to validate that all of the downstream clocks are ok when this is done though.

@tannewt Was something similar to making ```microcontroller.cpu.frequency``` settable ever done?  Running the stm32f405 at 168Mhz creates noise that jams GPS antennas, so the stm32f405 feather cannot be used with the GPS featherwing at that freq.  Having it settable would allow users that are using it with gps to change the frequency.  

..don't know if this is best as a continuation of this issue or if it should be opened as a new issue.

hierophect, 2021-03-04:
We haven't touched this in a while and currently the STM32 clock setup settings do not have any mechanism to change them at runtime. But I'd also note that allowing the user to lower clock speed across any microcontroller family could lead to some huge power improvements, now that we have the Alarm/DeepSleep module enabled.

dewalex, 2021-03-04:
> the STM32 clock setup settings do not have any mechanism to change them at runtime. 

Would it work putting an option in boot.py then?  (sorry if that's a dumb question, pretty new to this)

hierophect, 2021-03-04:
@dewalex no, unfortunately. There is no way to modify them from Python. If you are confident with C, the existing option for reducing clock speeds would be to build Circuitpython yourself, after redefining the clock macros in `ports/stm/peripherals/stm32f4/stm32f405xx/clocks.h` in `ports/stm/boards/<yourboard>/mpconfigboard.h`. This has been done once before for the `stm32f746g_discovery` board by an advanced user. You would also need to calculate the appropriate clock settings using ST's CubeMX tool or the reference manual, since it will not auto-calculate the USB clocks for you and they will break if not set to exactly 48MHz.

dewalex, 2021-03-05:
@hierophect sadly my confidence in C is even less than Python.  Do you think these boards could have two separate bin files for setting up CP, where the user could choose which one to install based off their use case?  I would imagine the GPS featherwing is the most popular wing, so there will be a lot of people that would have the need to run the clock one step down.

hierophect, 2021-03-05:
@dewalex I'd rather directly go for dynamic frequency adjustment through @tannewt's `microcontroller.cpu.frequency` suggestion, if possible. I'm not sure if we have precedent for having multiple bin files for different clock settings, it's possible we'd need to create a whole new board profile.

dewalex, 2021-03-05:
@hierophect Yes, that definitely sounds like the better route to go, I just wasn't sure how much work would be involved in that for getting it working for all boards.  The idea of clocking down to save power sounds really useful too.  Thanks for opening the new issue.

PR #2286 (2019-11-15): STM32: Raise clock speed of F405
This PR raises the clock speed of the F405 used in the feather_stm32f405_express and the pyboard to max value of 168MHz. Tested on a feather across a variety of modules. Resolves #2284. 

Thanks @manitou48 for bringing up and testing. 

Note to @dhalbert: us_delay dynamically calculates loop intervals for the no-interrupt case based on system frequency, so it's unaffected by this change. Double checked on the feather and we're still at ~1us accuracy.