adafruit/circuitpython #2292: How to read from mic on Edge Badge in CircuitPython?

scottalmond, 2019-11-16:
I have attempted to read data from the Adafruit Edge Badge mic in CircuitPython.  I do not see any templates in the provided example code on how to accomplish this.

After some digging, I believe I need "audiobusio.PDMIn"
https://circuitpython.readthedocs.io/en/4.x/shared-bindings/audiobusio/PDMIn.html
I tried the clk_pin as board.D5 and dat_pin as board.D6 per the silkscreen, but this generates an error.  The only viable combination of clk and data are D1 and D12.  All other combinations result in a "ValueError: Invalid clock pin" which I believe is being generated from here:
https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/audiobusio/PDMIn.c

This may be related to the following ticket.  If so, I would recommend putting a note in the Edge Badge documentation stating CircuitPythonn does not support the mic (also uploading a schematic of the Edge Badge would be appreciated too)
https://github.com/adafruit/circuitpython/issues/2187

scottalmond, 2019-11-16:
As a work-around I'm using a hardware hack leveraging the information from:
https://learn.adafruit.com/adafruit-pdm-microphone-breakout/circuitpython
Namely, I've set D5 and D6 to digital inputs, set D1 as CLK and D12 as DATA in the code, then routed jumpers from D1 --> D5 and D12 -->D6.  Using the PDMIn code in this configuration shows clear response to audio input (covering the mic significantly diminishes the RMS of the measurements).

tannewt, 2019-11-16:
Here is the schematic: [edge_badge.pdf](https://github.com/adafruit/circuitpython/files/3853942/edge_badge.pdf)

Looks like its attached to PA16 for clock and PA18 for data which are mapped correctly to D5 and D6.

Ok, I've found the Arduino implementation [here](https://github.com/adafruit/Adafruit_TFLite/blob/919865849bc18d6ce33dabe9d77f2a21136e5d70/examples/micro_speech_arcada/arduino_audio_provider.cpp#L44) and it's using the SERCOM in SPI mode to sample from the mic. CircuitPython's PDMIn can only use the I2S peripheral currently.

So, the fix would be to add support for using a SERCOM in SPI mode to read PDM data. The place to add it would be within PDMIn and just fallback to it if I2S isn't available.

tannewt, 2019-11-21:
@jepler This might be something you can look into.

ladyada, 2019-11-21:
@jepler will note this is very low priority - ill probably be redesigning the edgebadge, i was in a rush to get this rev done :) for now people can jumper to I2S pins

amithrishi, 2021-04-16:
Where is D1 on EdgeBadge?

amithrishi, 2021-04-16:
> As a work-around I'm using a hardware hack leveraging the information from:
> https://learn.adafruit.com/adafruit-pdm-microphone-breakout/circuitpython
> Namely, I've set D5 and D6 to digital inputs, [...]

I can't find D1? Please suggest.

scottalmond, 2021-04-16:
> I can't find D1? Please suggest.
Try the TX pin.  Please advise if that works or not; If not I'll have access to the old hardware in a few days and can check what I had setup at the time.

amithrishi, 2021-04-18:
> > I can't find D1? Please suggest.
> 
> Try the TX pin. Please advise if that works or not; If not I'll have access to the old hardware in a few days and can check what I had setup at the time.

I have already tried TX but the readings doesn't change if I speak into the microphone or not. Here is what I did. I jumped D12->D6 and D5->TX. I'm trying to read audio from the inbuilt microphone and I'm running this https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/PDM_Microphone/PDM_Mic_Sound_Level_Plot.py. I don't get any change in the values when I speak. I've tried speaking loudly and close to the mic as well.

Please suggest.

scottalmond, 2021-04-21:
Looking at my old code, I think you have to float the old pins too (set as input).  I've included some relevant parts below.  Please try this and let me know if you still encounter an error.  I can clean up what I have (was messily written during a hackathon) and just post the whole thing as a demo.  I'd also suggest blowing on the sensor as a means to test it, I found it was the lowest effort method to produce a lot of apparent noise (turbulence).

`            from adafruit_pybadger import PyBadger
import time
import math
import board
import digitalio
import audiobusio
import busio
import array
import random

pybadger = PyBadger()

pybadger.auto_dim_display(delay=30)

[...]

former_pin_clk=digitalio.DigitalInOut(board.D5) #D5 --> D1
former_pin_clk.switch_to_input(pull=None)
former_pin_dat=digitalio.DigitalInOut(board.D6) #D6 --> D12
former_pin_dat.switch_to_input(pull=None)

[...]

if(mic_init):
    mic=audiobusio.PDMIn(board.D1, board.D12, sample_rate=16000, bit_depth=16)
    mic_init=False

mic.record(buff, len(buff))
`

ECE884Group1, 2021-05-10:
> Looking at my old code, I think you have to float the old pins too (set as input). I've included some relevant parts below. Please try this and let me know if you still encounter an error. I can [...]

Thank you. It helped.

amitkrbhuyan, 2021-05-10:
# Main program
mic = audiobusio.PDMIn(board.TX, board.D12, sample_rate=16000, bit_depth=16)
samples = array.array('H', [0] * 160)
 
 
while True:
    mic.record(samples, len(samples))
    magnitude = normalized_rms(samples)
    print((magnitude,))
    time.sleep(0.1)

I don't understand. The sampling rate is 16000 but we store only 160 samples what happens to the rest. If I increase the number of samples, I don't see a rapid change on the serial plotter plotting the rms value. Please suggest.

amitkrbhuyan, 2021-05-10:
>
Thanks. It helped.

dhalbert, 2023-10-28:
Closing this because I think it's resolved.

closed 2023-10-28 by dhalbert