adafruit/circuitpython #2304: stm32f4: SPI clock rates

dhalbert, 2019-11-20:
@hierophect Could you check the baudrate calculation in `busio.SPI`? I was using an SPI clock to try to test the difference between the "bad" and "good" STM32 Feathers I have, and I got odd SPI clock rates on both:

| specified  | as measured on Saelae |
| ------------- | ------------- |
| 160 MHz | 20.83 MHz |
| 40 MHz | 10.42 MHz |
| 16 MHz |  5.208 MHz |
| 15 MHz | 5.208 MHz |
| 8 MHz  | 2.632 MHz  |
| 1 MHz | 328 kHz  |
| 100 kHz | 164 kHz|

hierophect, 2019-11-20:
@dhalbert Looks like everything is half of what it should be. I'll check it out. In terms of granularity, though, we aren't going to do much better - STM32 has a pretty limited divisor set for SPI. The actual values we're sorting from are:

**bus clock**
(168MHz / 2) => (PCLK2 = 84000000)

**SPI values:** 
/2 		42_000_000 
/4 		21_000_000
/8 		10_500_000
/16 		5_250_000
/32 		2_625_000
/64 		1_312_500
/128		656_250
/256	328_125

hierophect, 2019-11-20:
I thought all the SPI clocks were on the PCLK2 bus clock. They are not. Never trust what you read online, folks. Note that SPI2 and SPI3 will have half the maximum speed, since PCLK1 has a max speed of 42MHz.

PR #2307 (2019-11-21): STM32: SPI clock rate fix and cleanup
This PR corrects an issue in the STM32 port where an incorrect bus clock was used to calculate frequency on several instances of the SPI peripheral, halving the clock rate. The module now correctly selects a bus clock based on SPI instance. Thanks @dhalbert for finding this!

Since SPI was one of the first modules created for the port, I've also taken the opportunity to clean it up a bit to bring it more in line with the other modules. 

resolves #2304
closed 2019-11-21 by dhalbert