adafruit/circuitpython #2310: Need two calls to refresh() to update display

tomacorp, 2019-11-20:
Code is at https://github.com/tomacorp/edgemicscope
Tested versions of CircuitPython are 5.0.0 Beta 0 and 5.0.0 Alpha 5.
Hardware is the AdaFruit EDGE Badge, as distributed for the Hack-A-Day Superconference 2019.
After updating a lot of points on the display, two calls to refresh() are required:

    board.DISPLAY.refresh(minimum_frames_per_second=0)
    # Don't know why this second refresh is required.
    board.DISPLAY.refresh(minimum_frames_per_second=0)

To reproduce the problem, run the code and remove the second refresh() function.

The display updates properly with two calls.
The application is an oscilloscope for the digital microphone input.
It shows a trace moving across the screen with frequent updates.
Without the second call, a single trace is drawn and there are no more display updates.
A delay statement does not fix the problem.

I have not been able to reproduce the problem with a simpler example.
The example in the github repo code_simple.py is a simpler program that does not have the display refresh problem.

The full program is attached.
[code_py.txt](https://github.com/adafruit/circuitpython/files/3871318/code_py.txt)

DavePutz, 2019-11-26:
Investigating. In my testing with the provided code, if just above the 'while(True)' at line 141 an insert of
     board.DISPLAY.refresh(minimum_frames_per_second=0)
is done then only one call to refresh each time through the loop is needed. Still unsure why
the initial 'extra' call is needed.

tomacorp, 2019-11-27:
I tried moving the second call to line 141 before the while(True) and the
trace does not appear at all.
If you send me the whole program, I can test.

-Tom

DavePutz, 2019-11-28:
I have attached my test code. Note that it is using the light sensor rather than the mic since I didn't want to mess with the wiring to enable the mic. I was able to reproduce your issue using the light
sensor as input for the graph. You may need to tweak the adc_midscale at line 86 to adjust for your light levels.
[mytest_code_py.txt](https://github.com/adafruit/circuitpython/files/3899425/mytest_code_py.txt)

DavePutz, 2019-11-29:
Ran some more diagnostics. From the source code in shared-bindings/displayio/Display.c  I see: "If the call has taken too long since the last refresh call for the given target frame rate, then the refresh returns False immediately without updating the screen to hopefully help getting caught up." In the original test code checking the return from the call to board.DISPLAY.refresh() in the while loop I see that it returns True for the first call and False for every subsequent call (hence no display updates). Adding in a value for target_frames_per_second to the call ( 5 seemed like a good value); the return is now True most of the time and the display is getting updated. So, the issue here seems to be that the calls to refresh() were taking too long between calls, which is why an immediate second call updated the display. If you really can't run with auto_refresh True then it looks like you will need to tune target_frames_per_second to avoid having the display not update.

tomacorp, 2019-11-29:
Thanks! That is a great clue. It looks like this is working and solves the problem:
```
while not board.DISPLAY.refresh(minimum_frames_per_second=0):
    time.sleep(0.001)
```
I think this implements a blocking refresh.

closed 2019-11-29 by tomacorp

tannewt, 2019-12-02:
@tomacorp Did you try changing target_frames_per_second? It skips the refresh if you are behind the target rate so that the next frame will have a better chance of hitting it.

tomacorp, 2019-12-03:
@tannewt No, I did not try changing target_frames_per_second.

For this application, I think I need a blocking call to update the display.
Within this constraint, I want the display to update as frequently as
possible.
I am using this:
    while not board.DISPLAY.refresh(minimum_frames_per_second=0):
        pass

The idea is to do the erase, but not update the display until after the new
data is available.
Here is the pseudo-code -

while True:
    measure_waveform_data()
    draw_waveform_pixels()
    blocking_display_refresh()
    erase_all_the_waveform_pixels()

tannewt, 2019-12-03:
@tomacorp The refresh will almost always blocking if you set `target_frames_per_second=1` because your loop should always run faster than one frame per second.

tomacorp, 2019-12-03:
@tannewt The target_frames_per_second=1 did not work for me, because then
it only updates once a second. In some settings I like getting about 14
frames per second, which is a snappy scope!

I also tried minimum_frames_per_second=1 . When it first starts, the trace
does not appear. After about 1 second, I get "RuntimeError: Below minimum
frame rate".

tannewt, 2019-12-04:
Ah! Thanks for trying this. Looks like we should add support for `target_frames_per_second=None` to refresh every call.

kevinjwalters, 2020-03-09:
I have been trying to do a manual refesh and cannot find a combination that works on `5.0.0`. I've got some code that takes between 10-30ms roughly to do some stuff and update the Bitmap/etc and I'd then like to update LCD. There's no point updating it at 60Hz (16.667ms) as that'll end up drawing half an update some of the time.

If I just run `self._output.refresh(minimum_frames_per_second=0)` then the first one returns True and updates the screen and nothing after that updates the screen at the frequency I'm typically calling it at in one example loop.

If I specify `self._output.refresh(target_frames_per_second=1, minimum_frames_per_second=0)` then it works but it takes about a second per execution of `refresh()` which is a disaster. I tried 1000 and 500 for `target_frames_per_second` but that didn't help.

I'd like one that is a very simple refresh with no pauses. @tannewt Is that what your last suggestion of `target_frames_per_second=None` was? Does it make sense to re-open this?

tannewt, 2020-03-09:
@kevinjwalters I don't think we need to reopen this. It sounds like you want `target_frames_per_second=30` when your frames take up to 30ms to compute.

To reiterate how it works:
* `minimum_frames_per_second` dictates when refresh will raise an exception for being called too infrequently. `0` turns it off entirely. It doesn't impact whether a refresh actually occurs.
* `target_frames_per_second` dictates two things:
  1. If the time between the last refresh call and this one was greater than the time needed for the frame rate it will return without updating the display in order to speed up the next loop (and then hopefully refresh.) When a frame is skipped like this it isn't counted towards the minimum rate.
  2. If the time between the last refresh call and this one was shorter than the time needed for the target rate, it will sleep the remaining duration of the frame before refreshing. This is done so that game physics won't run faster than intended.

So, we could add `target_frames_per_second=None` for manual refresh but I think you'll get what you want with `target_frames_per_second=30` or somewhere between 1 that you said works and the default of 60 which doesn't.

kevinjwalters, 2020-03-10:
@tannewt Assuming you mean 33.3333ms? I can't exactly predict the time between frames so I'm not sure if that will always work, e.g. I think I'm back to no refreshes if it takes 35ms? Also, I'd like to do some useful processing in the application rather than have refresh() pause until a fixed time duration before returning.

kevinjwalters, 2020-03-10:
BTW, for games there's often acceptance that the game loop can take a variable amount of time. There's some features in MakeCode Arcade to support this and some are partially visible to application programmer, e.g. these [dx](https://arcade.makecode.com/reference/controller/dx) and [dy](https://arcade.makecode.com/reference/controller/dy) values scale with the duration of the game loop.

tannewt, 2020-03-10:
> @tannewt Assuming you mean 1000/30=33.3333ms? I can't exactly predict the time between frames so I'm not sure if that will always work, e.g. I think I'm back to no refreshes if it takes 35ms? As discussed previously, CP isn't predictable on timing. This seems like a needless risk?
> 
> Also, I'd like to do some useful processing in the application rather than have refresh() pause until a fixed time duration before returning.

Yes, correct. You'll be fine if it's only over the frame rate once in a while. It'll just drop a single frame in that case.

Feel free to open a new issue (and ideally a PR) to add `target_frames_per_second=None` support. It shouldn't be too hard.

> BTW, for the use case of games there's often acceptance that the game loop can take a variable amount of time. There's some features in MakeCode Arcade to support this and some are partially visible to application programmer, e.g. these [dx](https://arcade.makecode.com/reference/controller/dx) and [dy](https://arcade.makecode.com/reference/controller/dy) values scale with the duration of the game loop.

Ya, I know another option is to pass the duration into the math but I decided against that for now because it makes the math more complex.