adafruit/circuitpython #2332: nrf52840 feather: crash on file I/O

jepler, 2019-11-27:
Create a file `test.bin` on your filesystem of 4096 bytes, and execute this e.g., in the repl:
```
f = open("test.bin", "rb")
f.seek(2048)
b = bytearray(2048)
v = memoryview(b)
f.readinto(v[909:])
```
circuitpython will crash.  The underlying cause seems to be that a non-aligned buffer is sent to `nrfx_qspi_read`, which is not supported and triggers an assertion:
```
#1  0x00052fc0 in nrfx_qspi_read (p_rx_buffer=p_rx_buffer@entry=0x20008cdd, 
    rx_buffer_length=rx_buffer_length@entry=512, 
    src_address=src_address@entry=33280) at nrfx/drivers/src/nrfx_qspi.c:297
297	    NRFX_ASSERT(nrfx_is_word_aligned(p_rx_buffer));
```

Full traceback at the site the assertion is raised:
```
#0  __assert_func (file=file@entry=0x7ed38 "nrfx/drivers/src/nrfx_qspi.c", 
    line=line@entry=297, 
    func=func@entry=0x7ecc8 <__FUNCTION__.8276> "nrfx_qspi_read", 
    expr=expr@entry=0x7ed88 "nrfx_is_word_aligned(p_rx_buffer)")
    at ../../main.c:508
#1  0x00052fc0 in nrfx_qspi_read (p_rx_buffer=p_rx_buffer@entry=0x20008cdd, 
    rx_buffer_length=rx_buffer_length@entry=512, 
    src_address=src_address@entry=33280) at nrfx/drivers/src/nrfx_qspi.c:297
#2  0x000487f2 in spi_flash_read_data (address=address@entry=33280, 
    data=data@entry=0x20008cdd "", length=length@entry=512)
    at supervisor/qspi_flash.c:90
#3  0x00047f6c in read_flash (address=33280, data=data@entry=0x20008cdd "", 
    data_length=data_length@entry=512)
    at ../../supervisor/shared/external_flash/external_flash.c:82
#4  0x00048574 in external_flash_read_block (dest=dest@entry=0x20008cdd "", 
    block=block@entry=65)
    at ../../supervisor/shared/external_flash/external_flash.c:493
#5  0x000486b8 in supervisor_flash_read_blocks (dest=0x20008cdd "", 
    block_num=65, num_blocks=2)
    at ../../supervisor/shared/external_flash/external_flash.c:545
#6  0x00043f42 in flash_read_blocks (dest=0x20008cdd "", 
    block_num=<optimized out>, num_blocks=<optimized out>)
    at ../../supervisor/shared/flash.c:110
#7  0x00042e5a in disk_read (pdrv=<optimized out>, 
    buff=buff@entry=0x20008cdd "", sector=sector@entry=66, count=count@entry=2)
    at ../../extmod/vfs_fat_diskio.c:74
#8  0x0004d346 in f_read (fp=0x20008734, buff=<optimized out>, btr=1139, 
    br=br@entry=0x2003fda4) at ../../lib/oofatfs/ff.c:3389
#9  0x00043122 in file_obj_read (self_in=<optimized out>, buf=<optimized out>, 
    size=<optimized out>, errcode=0x2003fdd8) at ../../extmod/vfs_fat_file.c:71
#10 0x0003d5e4 in mp_stream_rw (stream=0x20008730, buf_=<optimized out>, 
    size=1139, errcode=errcode@entry=0x2003fdd8, flags=flags@entry=0 '\000')
    at ../../py/stream.c:63
#11 0x0003d7b2 in stream_readinto (n_args=2, args=0x2003fedc)
    at ../../py/stream.c:298
#12 0x00036560 in fun_builtin_var_call (
    self_in=0x75e08 <mp_stream_readinto_obj>, n_args=2, n_kw=0, 
    args=0x2003fedc) at ../../py/objfun.c:127
#13 0x00031d8e in mp_call_function_n_kw (
    fun_in=0x75e08 <mp_stream_readinto_obj>, n_args=2, n_kw=0, 
    args=args@entry=0x2003fedc) at ../../py/runtime.c:622
#14 0x00031e58 in mp_call_method_n_kw (n_args=<optimized out>, 
    n_kw=<optimized out>, args=args@entry=0x2003fed8) at ../../py/runtime.c:637
#15 0x0003f60e in mp_execute_bytecode (code_state=code_state@entry=0x2003fec0, 
    inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1015
#16 0x00036608 in fun_bc_call (self_in=0x20009270, n_args=<optimized out>, 
    n_kw=204175, args=0x0) at ../../py/objfun.c:284
#17 0x00031d8e in mp_call_function_n_kw (fun_in=fun_in@entry=0x20009270, 
    n_args=n_args@entry=0, n_kw=n_kw@entry=0, args=args@entry=0x0)
    at ../../py/runtime.c:622
#18 0x00031db8 in mp_call_function_0 (fun=fun@entry=0x20009270)
    at ../../py/runtime.c:596
#19 0x0004eb78 in parse_compile_execute (source=source@entry=0x2003ffb8, 
    input_kind=<optimized out>, exec_flags=exec_flags@entry=22, 
    result=result@entry=0x0) at ../../lib/utils/pyexec.c:114
#20 0x0004ee3e in pyexec_friendly_repl () at ../../lib/utils/pyexec.c:518
#21 0x00043804 in run_repl () at ../../main.c:401
#22 0x00043886 in main () at ../../main.c:457
```

jepler, 2019-11-29:
This also requires that the test.bin file be contiguous on disk; you may need to initialize the filesystem before creating test.bin and running the reproducer steps.

ladyada, 2019-12-10:
also tagging @hathach FYI since this could happen in arduino :)

jepler, 2019-12-10:
(based on discord discussion) For now we are going to take the band-aid fix but @tannewt wants to avoid the copy that this introduces by handling it in `nrfx_qspi_read` (or maybe `spi_flash_read_data`) for nRF: when the input buffer is unaligned, do a (1, 2, or 3-byte) read to a 4-byte, aligned region on the stack, and copy enough bytes from there to re-align the data pointer.  Then do the bulk of the read as now, avoiding any extra copies.

hathach, 2019-12-11:
@ladyada  yeah, right, it could be an possible issue with arduino as well. `nrfx_qspi_write` also require memory to be 4 bytes aligned as well. I will file an issue on Arduino SPIFLash repo here https://github.com/adafruit/Adafruit_SPIFlash/issues/40

commit a484a93b (2019-12-10): nRF: disk_read must be 4-byte aligned

.. a requirement that oofatfs needs to be taught to respect.

This problem can be demonstrated with the following snippet, except
that the related file ("test.bin") must also be contiguous on the
filesystem.  You can ensure this by reformatting your device's filesystem
before testing, then copying any single file bigger than 4kB to test.bin.

    f = open("test.bin", "rb")
    f.seek(2048)
    b = bytearray(2048)
    v = memoryview(b)
    f.readinto(v[909:])

Closes: #2332
commit eef742bf (2020-03-03): oofatfs: Remove _FS_DISK_READ_ALIGNED

This workaround is no longer needed, so it can be removed.

Closes: #2332
closed 2020-03-03 by tannewt