adafruit/circuitpython #2334: Increase monotonic_ns precision

olivier-boesch, 2019-11-28:
Hi Everyone,

I'm using M4 express (circuitpython v. with circuitpython in physics and geophysics education. I'm on a project that measure the speed of wave in rocks.

I was afraid, when i saw the precision of the monotonic_ns function as 1ms only.

Is it possible to make it work in a more precise way like the micros function in arduino ?

Thanks.

Olivier B.

ladyada, 2019-11-28:
please use pulsein for precision timing, Python is not a real-time precision programming language
https://circuitpython.readthedocs.io/en/latest/shared-bindings/pulseio/PulseIn.html

olivier-boesch, 2019-11-29:
As I'm used two analogic inputs and measuring between these two signals, using pulsein is not possible. Is it ?

If I convert my schematic to use only one input, I could use pulsein but what is the minimum level to get a high with pulsein?

Thanks

ladyada, 2019-11-29:
please post a link to your code/reference to the project so we can see what you are trying to do

PR #2601 (2020-02-07): Monotonic ns improvement; time.sleep rounds to nearest msec
Fixes #2334. Fixes #2508.

- Increase precision of `time.monotonic_ns()` to microseconds on all ports. I need some rough timing values that were more granular than milliseconds.
- Round `time.sleep()` to the nearest millisecond. Tested #2508 sample program and now get 100 Hz as intended.

`time.monotonic_ns()` and `time.sleep()` should still not be used for precise timing, but this makes them more useful than they were before.

Note that `time.monotonic()` still increases only on milliseconds. It was not worth improving due to the low [... cut]
closed 2020-02-07 by tannewt