adafruit/circuitpython #2335: `os.open` creates file with wrong `mtime`

iot49, 2019-11-28:
Files created with `os.open` have an incorrect `mtime`. The code below verifies that the time is set correctly (output `now`), then creates a file and prints the `mtime`, which is always in Dec 2016, regardless of the current time.
Hardware: nrf52840 particle Xenon with CircuitPython 5.0 beta.

```python
import os
import time

def current_time(lt = time.localtime()):
    return "{:2d}-{:02d}-{:02d} {:2d}:{:02d}:{:02d}".format(lt.tm_year, lt.tm_mon, lt.tm_mday, lt.tm_hour, lt.tm_min, lt.tm_sec)

print("now:  ", current_time())

file = '/fs_test/mcu.txt'
try:
    os.remove(file)
except:
    pass

with open(file, 'w') as f:
    f.write("hi from MCU")
    
print("mtime:", current_time(time.localtime(os.stat(file)[7])))

# output:
# now:   2019-11-28 21:49:40
# mtime: 2016-12-04  0:18:22
```

tannewt, 2019-12-03:
This is due to nrf not implementing `get_fattime`: https://github.com/adafruit/circuitpython/blob/master/ports/nrf/fatfs_port.c#L31

The fix would be to factor `get_fattime` out and make it depend on `rtc_get_time_source_time` which [is used by `localtime`](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/time/__init__.c#L236)

For reference, SAMD [uses `common_hal_rtc_get_time`](https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/fatfs_port.c#L40) which will only work for the internal RTC I believe instead of any time source.

iot49, 2019-12-09:
The SAMD solution works "perfect" (at least with the internal RTC which is what I am using). Would you like a PR?

I've also tried the following code:

```python
    timeutils_struct_time_t tm;
    struct_time_to_tm(rtc_get_time_source_time(), &tm);
    mp_uint_t secs = timeutils_seconds_since_epoch(tm.tm_year, tm.tm_mon, tm.tm_mday,
                                                  tm.tm_hour, tm.tm_min, tm.tm_sec);

```

Then files cannot be written anymore, not even `boot_out.txt` is created.

tannewt, 2019-12-09:
@iot49 a PR would be awesome!

dhalbert, 2020-02-06:
Fixed by #2367

closed 2020-02-06 by dhalbert