adafruit/circuitpython #2339: pca10059 file system issues with CP 5.0-beta.0

jerryneedell, 2019-11-30:
Just a heads up that I have run into serious issues with the current master CP build for the pca10059.

Attempting to erase files from the file system or executing storage.erase_filesystem() causes the board to hand be completely unusable --and caused my linux box to hang!!
I reproduced this on 2 different pca10059's
I had to reflash the bootloader to recover the board -- it works fin with CP 4.1.0

I'll do more testing tomorrow, but wanted to put out a warning.

I seemed to work OK at first when just loading the latest build - I was able to execute files already on the board. The problems occurred when I tried to erase old library files to update them.

jerryneedell, 2019-11-30:
Test case:
Reflash bootlader -- from Adafruit_nRF52_Bootloader
make BOARD = pca10059 erase
make BOARD = pca10059 sd
make BOARD = pca10059 flash

load CP 4.1.0 to board
copy " blinky.py" to board
all OK -- can erase/rcopy blinky.py
leave blinky.py on board

load CP5.0 (current master build)
looks OK -- blinky.py executive OK via REPL
 delete blinky. py

PCA10059 hangs -- eventually disconnects from USB

reboot - single reset fails -- can connect to USB but no REPL

double RESET does not work
attempted to run "df" on linux box -- linux box hung for a few minutes - evetyall responded to control C

here are the dimes logs:

jerryneedell, 2019-11-30:
I tried using a Maker Diary nrf52840 USB dongle and am having the same issues....
perhaps it is common to the nRF52840 builds without an SPI flash file system...

dhalbert, 2019-11-30:
I am revamping the internal flash partitioning and am suspicious that there’s an overlap between `nvm` and the file system. But it might be different problem too.

dhalbert, 2019-12-11:
I revamped the internal flash layout, and I am still seeing some problems. CIRCUITPY will appear but then the board will crash. This does not sounds like your symptoms.

I do NOT see a problem if I compile with `DEBUG=1`, so it may be some kind of timing problem. Sometimes when I switch back to a non-debug build it will work for one or two times and then fail. There is no difference in the flash layout between the debug and non-debug builds, except that the firmware is bigger because it's not optimized.

jerryneedell, 2019-12-11:
ON my PCA10059 -- I can load 5.0 Beta.1 and even execute a program that is stored on it. It is only when I try to delete a file from the board that it crashes -- hangs then disconnects
I can refresh the .uf2 file if I just do a doubletap after trying to delete one file.
 
I have not repeated the storage.erase_filesystem() test when it requires the J-Link to recover....

jro22, 2019-12-11:
Last evening I built and flashed a uf2 bootloader onto an Arduino nano 33 ble board, hoping to try out CP 5.0 Beta.1.  Unfortunately I think I am running into this issue as well.  

After a double reset press the boot disk (NANO33BOOT) appears and I can successfully drag the CP .uf2 file to it.  When the board resets, CIRCUITPY appears to mount.  At this point If I'm quick I can open code.py in the mu editor, but any attempts to edit or save it fail with the board hanging and then disconnecting.  No luck getting REPL to run either.

Here's a quick dump of what I was able to do from the terminal between when CIRCUITPY mounted and when it disappeared:
```
bad-sector:~ jolson$ df /Volumes/CIRCUITPY
Filesystem   512-blocks Used Available Capacity iused ifree %iused  Mounted on
/dev/disk2s1        477    4       473     1%     512     0  100%   /Volumes/CIRCUITPY
bad-sector:~ jolson$ ls /Volumes/CIRCUITPY/
boot_out.txt	code.py		lib
bad-sector:~ jolson$ cat /Volumes/CIRCUITPY/boot_out.txt 
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Arduino Nano 33 BLE with nRF52840
bad-sector:~ jolson$ cat /Volumes/CIRCUITPY/code.py 
cat: /Volumes/CIRCUITPY/code.py: Device not configured
```

PR #2376 (2019-12-12): nrf: sd_flash_operation_status should be volatile
A flag variable, `sd_flash_operation_status`, used to check when an internal flash operation was done was not `volatile`, so it could not reliably be checked.
```
STATIC sd_flash_operation_status_t sd_flash_operation_wait_until_done(void) {
    while (sd_flash_operation_status == SD_FLASH_OPERATION_IN_PROGRESS) {
        sd_app_evt_wait();
    }
    return sd_flash_operation_status;
}
```
Fixes #2339, I believe.

I think maybe this broke with better inlining and inter-routine optimization after the switch to gcc9. I looked at the assembly code  [... cut]
closed 2019-12-12 by tannewt