adafruit/circuitpython #2341: NeoPixels do not turn off on Circuit Playground Bluefruit on REPL soft reboot

kevinjwalters, 2019-11-30:
There's some difference in behaviour here between the Circuit Playground Bluefruit and Circuit Playground Express. If you turn on some pixels and then hit Cntrl-D in REPL on a CPX then pixels turn off. On a CPB it they remain on.

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.0 on 2019-11-19; Adafruit Circuit Playground Bluefruit with nRF52840
>>>
>>> from adafruit_circuitplayground.bluefruit import cpb
>>> cpb.pixels[5] = (50, 0, 0)
>>> cpb.pixels[6] = (0, 50, 0)
>>> cpb.pixels[7] = (0, 0, 50)
>>>
soft reboot
```

This is on `5.0.0-beta.0` with the latest library copied in from https://github.com/adafruit/Adafruit_CircuitPython_CircuitPlayground

dhalbert, 2019-12-17:
CPX resets the neopixels to all off in board.c:reset_board(). Do the same for CPB.

kevinjwalters, 2020-01-18:
It wasn't in the release notes but I thought I'd check `5.0.0.beta.3`. FYI, I'm still seeing the NeoPixels left on.

PR #2548 (2020-01-24): reset NeoPixels on soft reload on CPB, pybadge, and pygamer boards 
Fixes #2341. Soft-reload was not turning off NeoPixels on Circuit Playground Bluefruit.

- Add code to CPB, pygamer boards, and pybadge boards `board.c` to turn off NeoPixels.
- Refactor the port-independent NeoPixel reset code to avoid code duplication. Add `supervisor/shared/board.c` for such common code.
- Allocate space on the stack for up to 24 NeoPixels in nrf `neopixel_write/__init__.c` to enable PWM-based NeoPixel writes without a VM but with the SoftDevice enabled. I chose 24 because it's less than 400 bytes on the st [... cut]
closed 2020-01-24 by tannewt