adafruit/circuitpython #2347: Bluefruit code crashing to Hardfault_Handler

kattni, 2019-12-03:
Running 5.0.0.beta0 on Circuit Playground Bluefruit. This code is adapted from the Central example in the Adafruit CircuitPython BLE repo. The commented out part is sending button packets without debouncing (for project testing purposes) - it crashes regardless of which code I'm using. Current code.py attached.

[hardfault_handler_crash.py.zip](https://github.com/adafruit/circuitpython/files/3918242/hardfault_handler_crash.py.zip)

dhalbert, 2019-12-03:
Do you have to push the buttons in some way to initiate the crash?

kattni, 2019-12-03:
No, it crashes with no interaction.

dhalbert, 2019-12-04:
Here's the code for the other side: 
[CPB_client.zip](https://github.com/adafruit/circuitpython/files/3922135/CPB_client.zip)

I was unable to get this to crash with the original code: I ran it for several hours. I originally thought I had reproduced it by removing the `sleep` on the central side, but a second try didn't produce that. I had other troubles (USB freezing up on Ubuntu) and had to reboot, but they may have been unrelated. After rebooting it was fine.

kattni, 2019-12-04:
My updated code ran all night without crashing. I copied a file off of the board today and it crashed into the `Hardfault_Handler`. 

@dhalbert I set it up on a different board that is also incidentally the new rev. I copied all the necessary files to it, and it immediately crashed into the `Hardfault_Handler`. On reset, it came back up and seems to be working so far.

Including the current code with libraries.

[CPB_central.zip](https://github.com/adafruit/circuitpython/files/3923409/CPB_central.zip)
[CPB_client.zip](https://github.com/adafruit/circuitpython/files/3923410/CPB_client.zip)

kattni, 2019-12-04:
Another update: every time I write something to the client, it soft reloads both boards, and fairly consistently crashes the central into safe mode with the `Hardfault_Handler`. I'm on MacOS.

tannewt, 2019-12-05:
I saw this a bunch with the RGB LED. Here is the backtrace and then the backtrace after breaking when given the bad input. The cause is that when discovering the services available over a connection we're not handling the case where the discovery is terminated by disconnect. In that case, the service list to tuple code will read the internal connection pointer which is NULL and then attempt to convert the data at the list offset to a tuple. Fixed in https://github.com/tannewt/circuitpython/tree/fix_service_list_crash which will raise a ConnectionError when it happens.

```
Breakpoint 1, reset_into_safe_mode (reason=reason@entry=HARD_CRASH) at ../../supervisor/shared/safe_mode.c:85
85	void __attribute__((noinline,)) reset_into_safe_mode(safe_mode_t reason) {
(gdb) bt
#0  reset_into_safe_mode (reason=reason@entry=HARD_CRASH) at ../../supervisor/shared/safe_mode.c:85
#1  0x00043860 in HardFault_Handler () at supervisor/port.c:168
#2  <signal handler called>
#3  service_linked_list_to_tuple (services=0xa81) at ../../shared-bindings/_bleio/Service.c:192
#4  0x00000000 in ?? ()
```

```
Program received signal SIGTRAP, Trace/breakpoint trap.
service_linked_list_to_tuple (services=0xa81) at ../../shared-bindings/_bleio/Service.c:188
188	        asm("bkpt");
(gdb) bt
#0  service_linked_list_to_tuple (services=0xa81) at ../../shared-bindings/_bleio/Service.c:188
#1  0x00058232 in common_hal_bleio_connection_discover_remote_services (self=self@entry=0x2002b6e0, service_uuids_whitelist=<optimized out>)
    at common-hal/_bleio/Connection.c:663
#2  0x00053632 in bleio_connection_discover_remote_services (n_args=<optimized out>, pos_args=<optimized out>, kw_args=<optimized out>)
    at ../../shared-bindings/_bleio/Connection.c:155
#3  0x000364da in fun_builtin_var_call (self_in=0x7a9a0 <bleio_connection_discover_remote_services_obj>, n_args=2, n_kw=0, args=0x2003fc70)
    at ../../py/objfun.c:122
#4  0x00031d02 in mp_call_function_n_kw (fun_in=0x7a9a0 <bleio_connection_discover_remote_services_obj>, n_args=2, n_kw=0,
    args=args@entry=0x2003fc70) at ../../py/runtime.c:622
#5  0x00031dca in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x2003fc6c) at ../../py/runtime.c:637
#6  0x0003f3ee in mp_execute_bytecode (code_state=code_state@entry=0x2003fc58, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#7  0x0003655c in fun_bc_call (self_in=0x200369d0, n_args=<optimized out>, n_kw=204035, args=0x2003fd68) at ../../py/objfun.c:284
#8  0x00031d02 in mp_call_function_n_kw (fun_in=0x200369d0, n_args=2, n_kw=0, args=args@entry=0x2003fd68) at ../../py/runtime.c:622
#9  0x00031dca in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x2003fd64) at ../../py/runtime.c:637
#10 0x0003f3ee in mp_execute_bytecode (code_state=code_state@entry=0x2003fd50, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#11 0x0003655c in fun_bc_call (self_in=0x20036a40, n_args=<optimized out>, n_kw=204035, args=0x2003fdd8) at ../../py/objfun.c:284
#12 0x00031d02 in mp_call_function_n_kw (fun_in=0x20036a40, n_args=2, n_kw=n_kw@entry=0, args=args@entry=0x2003fdd8) at ../../py/runtime.c:622
#13 0x0003c22a in instance_subscr (self_in=0x2002b7a0, index=0x20035ba0, value=0x8) at ../../py/objtype.c:859
#14 0x00033dbc in mp_obj_subscr (base=0x2002b7a0, index=0x20035ba0, value=value@entry=0x8) at ../../py/obj.c:493
#15 0x0003fc0a in mp_execute_bytecode (code_state=code_state@entry=0x2003fe90, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:388
#16 0x0003655c in fun_bc_call (self_in=0x20039720, n_args=<optimized out>, n_kw=204035, args=0x0) at ../../py/objfun.c:284
#17 0x00031d02 in mp_call_function_n_kw (fun_in=fun_in@entry=0x20039720, n_args=n_args@entry=0, n_kw=n_kw@entry=0, args=args@entry=0x0)
    at ../../py/runtime.c:622
#18 0x00031d2c in mp_call_function_0 (fun=fun@entry=0x20039720) at ../../py/runtime.c:596
#19 0x0004e9de in parse_compile_execute (source=source@entry=0x731a0, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT,
    exec_flags=exec_flags@entry=32, result=result@entry=0x2003ffbc) at ../../lib/utils/pyexec.c:114
#20 0x0004ed1c in pyexec_file (filename=filename@entry=0x731a0 "code.py", result=result@entry=0x2003ffbc) at ../../lib/utils/pyexec.c:529
#21 0x00043312 in maybe_run_list (filenames=filenames@entry=0x20004038 <supported_filenames>, exec_result=exec_result@entry=0x2003ffbc)
    at ../../main.c:183
#22 0x000433e2 in run_code_py (safe_mode=safe_mode@entry=NO_SAFE_MODE) at ../../main.c:241
#23 0x000436d2 in main () at ../../main.c:464
```

kattni, 2019-12-06:
Tested UF2 posted by @tannewt in Discord with a fix and so far I have not run into the `Hardfault_Handler` again.

PR #2362 (2019-12-07): Check connection validity after service discovery.
Instead of HardFaulting it will now raise a ConnectionError when a disconnect happens while discovering services.

Fixes #2347
commit da0ea979 (2019-12-07): Check connection validity after service discovery.

Fixes #2347
closed 2019-12-07 by dhalbert