adafruit/circuitpython #2354: busio.UART reports not supporting 9 bit bata

ericboxer, 2019-12-04:
Based on the [Documentation](https://circuitpython.readthedocs.io/en/4.x/shared-bindings/busio/UART.html) busio.UART should support 9 bit data. However, when i try to create a UART port 

```python
serial485 = busio.UART(board.TX, board.RX, baudrate=38400, bits=9)
``` 
I get the following NotImplimentedError:
```
NotImplementedError: bytes > 8 bits not supported
```

hierophect, 2019-12-04:
From what I can tell, the only Circuitpython devices that will support variable UART word length are the STM32s, and even they can only do 8 or 9, not 7. Atmel, NRF and CXD all either throw errors or ignore it.

dhalbert, 2019-12-04:
We dropped support for 9 bits for atmel-samd due to complications when upgrading from the atmel-samd ASF3 to ASF4 libraries. The documentation should be clearer.

@hierophect I do see that stm32f4 supports 9 bits, but the read/write buffering seems to assume max 8-bit bytes. Maybe this is a bug? We need to decide whether it's worth supporting 9-bit bytes.

@ericboxer What is your application that needs 9-bit bytes?

ericboxer, 2019-12-04:
@dhalbert Gotcha, that makes sense.

I needed a quick and dirty to connect up to an existing RS485 bus that uses 9 bit. The main device on the bus is an STM32F4, and one of the other slave devices is an older SiLabs 8051.

hierophect, 2019-12-05:
@dhalbert the 9th bit would be handled by the peripheral as part of the parity system, so it isn't actually part of the data. I assume when parity is used in 8 bit mode it simply disregards the 8th bit of data if you provide it.

dhalbert, 2019-12-05:
@hierophect Looks like RS485 can use the 9th (formerly parity) bit to distinguish between data and command (eek). On SAMD, you can send 9 bits of data plus a parity bit plus a stop bit. 9-bit SPI is also available. So it can be a real data bit.

hierophect, 2019-12-05:
@dhalbert I looked into it more and while it appears at first glance that you'd need to use the LLs to properly handle 9 bit data (they have specific 9 bit send and receive functions) it actually looks like the ST peripheral will simply spread the data out across two bytes for every send and receive when using 9 bit data without parity. From the user's perspective they'll appear to receive and send twice as much data, they just have to manage that conversion themselves.

hierophect, 2019-12-05:
I'm not sure what the python equivalent of this would be but if it was in C you'd simply cast the 16 bit array holding your 9 bit words as 8 bit and it should work ok. 
https://community.st.com/s/question/0D50X00009XkWx7SAF/9bit-uart-transmission-stm32-cubemx-hal-driver